0xGiddi/qnap8528 #2: Led control

jukrut, 2024-10-28:
Hello and thank you for you work,
fan and temp work very well.

I am very interested in getting the LED control feature to work.

what I saw in your comments:
> front-panel:green:disk1N

wouldn't it be easier to have 
> front-panel:disk1N and set that to green, red, alternate? 

i have 0 xp in that kind of development, but if you need someone to test or try out different values or so... I'm very much willing to do so.

0xGiddi, 2024-10-28:
I have been meaning to get the LEDs working for a while but I have been
swamped with other work,  since I want the driver to be universal, I have
been having trouble structuring the code properly since different devices
have different LEDs.

The are a number of reasons why the LEDs are structured like that. The
first is that the Linux led subsystem does not support bi-color LEDs. Each
led function and color has its own instance, there is support for
multicoloured LEDs however the support extends to RGB values and not
specific bi-color (more mixing channels), I can add a custom attribute to
each led to indicate color, I have not decided to take that approach since
I am trying to be as generic as possible ( that's the whole reason for a
kernel module and standard interfaces, so a userland program can detect all
the devices without special code)

Another problem with the LEDs is the blinking, take for example the status
LED, it can blink either color or both and that creates a state dependency
between the two led instances (they are back to back diodes, cannot be on
together, as far as I have determined).

There is also the another small problem that setting an led does not seem
to overwrite the current state, so an led set to blink green on a disk (for
example) will not blink if it's set on error or similar (can't remember
exactly) so the core will need to know the current led state / revert all
to off and then turn them back on, I also already have a basic idea of how
to implement this, just waiting to write and test it properly.

Last annoyance is the fact that the LEDs are marked differently, on
different devices (internal EC indexes) so it's not as simple as disk 1
being EC index 1, for example m.2 1 on the x73A is actually EC index 9. I
am working a structure to hold specific configs and apply the correct one
depending on the backplane and motherboard code from the vpd so that this
module could be used with as many qnap devices that use the 8528 EC.

I will try and get around to the LEDs, even if it's a simple solution for
now in the next two weeks, if you have any better ideas or want to
contribute feel free ( you can even do it via a ticket on GitHub so other
people can know).

Thank you for offering to test, I will make sure to let you know if I have
anything specific. What model do you have?

jukrut, 2024-10-28:
:) ah I see.. thank you for explaining it in such detail.

about multicolor, do you mean that: https://docs.kernel.org/leds/leds-class-multicolor.html ?

I hold back with the ideas, because I have so little understanding about it :)

https://www.kernel.org/doc/html/v5.8/leds/leds-class.html

I'll read through that too...

i have a QNAP TS-473A-8G. ( what i read, you also have that )

>so the core will need to know the current led state / revert all
to off and then turn them back on

this i don't understand.
Why does it need to know the state?

e.g. hdd1 to green

if you want to set it to light green,.. reset red, green, blink,.. and enable front-panel:green:disk1.

There is the HDD activity shown with a blinking green light,.. would the setting overwrite that?
If so would disable all then show the activity again with blinking light?

Thank you up front for explaining :)
With that less knowledge it's even hard to find the proper words, to ask a question :)

0xGiddi, 2024-10-28:
> about multicolor, do you mean that: https://docs.kernel.org/leds/leds-class-multicolor.html ?
Yes, exactly that. 
> I hold back with the ideas, because I have so little understanding about it :)

No problem, feel free to comment. 
> https://www.kernel.org/doc/html/v5.8/leds/leds-class.html
> 
> I'll read through that too...

That's the generic led subsystem that I am planning on using. 

> > so the core will need to know the current led state / revert all
> > to off and then turn them back on
> 
> this i don't understand. Why does it need to know the state?
> 
> e.g. hdd1 to green
> 
> if you want to set it to light green,.. reset red, green, blink,.. and enable front-panel:green:disk1.

For the HDDs its a little less complicated, as, yes, we can turn off all the states and then select the state we want, the problem is mainly with the STATUS LED. If I expose two LEDs for status, green and red, the I can turn on and blink the two colors, however, how do I represent the state of blinking red-green in an alternating pattern since I have no multicolored LED? (in the EC, the status LED "blink" state has 3 options, green, red, both), My intention is to implement the LEDs as separate colors and if you set the red and green status LED to blink then it will blink both, but if I dont save the state and turn off all LEDs before setting my new state, I will loose that functionality. Its not a real big problem, but I would like to keep the interface as intuative as possible. Another example is that IDENT overrides the disk LED statuses, I will figure a nice way to expose this (maybe not expose IDENT at all?)

> There is the HDD activity shown with a blinking green light,.. would the setting overwrite that? If so would disable all then show the activity again with blinking light?

Yes, there is a potential to break the activity blinking by getting the LED states out of sync somehow, I have done this a fwe times when running test code while reverse engineering, I have not yes documented all edge cases and will do that if needed when I implement the LEDs.

*Now, allow me to go off on a tangent:* Don't get me started of the horrible implementation of the disk activity LEDs on the QNAP devices, did you install a different OS yet? On my machine I noticed the two M.2 leds do not blink with activity and stay on solid green, the first two  HDD bays on the left  turn on and blink green with disk activity (as your would expect), but after a while I noticed that the two second disks (on the right) never blink and stay on solid. At first I thought it was due to my ZFS setup being in RaidZ2, which maybe caused more activity on the first two disks that are the parity, but no, the activity LEDs on the two right disks do not work out of the box. Why? I think it's something to do with the hardware architecture of the device that has those specific two SATA ports connected to the main board via a PCI bridge.

I thought of adding a whole new section to the kernel module to hook the block IO operations, resolve the correct LED according to PCI/DISK paths and blink it, but I will not be doing this since:
1.  Changes added to the Linux kernel (5.10+ I think) that makes dynamic hacking a pain (no kallsyms lookup and more).
2. I have no idea how this behaves on the 673A or the 873A or other devices that mau be supported, so every device will need to be tested and a more complicated config config added. 
3. There is no good way to resolve a physical SATA disk slot to an EC LED index/Physical disk slot (I thought to use the PCI BDF, but again, this will need to be added manually as a dictionary, but then we get into the world of enumeration order, what happens if a PCI card is added (GPU,NVME etc.) it might cause the enumeration order to change during boot, and then the PCI BDF maybe wrong). 
4. I want the kernel module to be as simple as possible and not interfere with critical OS operations to minimize risks of data and performance loss. 

I might just write a userland script that will slow-monitor block IO and trigger the LEDs manually once exposed properly via the module, maybe use a system-tap on the block IO? I have no idea how I would like to implement this yet.  
*/rant* 

> Thank you up front for explaining :) With that less knowledge it's even hard to find the proper words, to ask a question :)
How else do we learn? :)

jukrut, 2024-10-29:
As simple as possible sounds perfect.

Maybe from another view point, what would one like to do with the leds?
- if e.g. a should get replaced, I would like to identify it
- if e.g. a disk has errors...  (smart, zfs.. whatever), I would like to show that..

(but before continuing what I would like to do a question)

> There is no good way to resolve a physical SATA disk slot to ....

about that:
If I understand correctly.. it is hard to find out which LED ID belongs to e.g. /dev/sda or /dev/disk/by-id correct ?

that is probably fine for a kernel module, and the userland tools using it, have to know which drive is installed in which slot... so for e.g. sda blink led0.
I anyway thought about documenting the serial of the hdd  that i put in the slots.. so that I already know somewhere. And this config could then be used to blink sda LED..

about custom OS.

Yes i first had debian, currenlty ubuntu.

And I also had that problem, that 2 leds did nothing after a while..
but without changing anything, last time when replaced a disk it worked.
( with: `dd if=/dev/sdX of=/dev/null bs=1M count=5000` ).
But that won't work well, if all disks are busy.. at home I disabled smbd and others and nothing happened on the other drives.

IDENT is not so important for me, because I have 1 NAS :)

In general I will not rely that much on the LED for monitoring the health or alerting, because I use other tools for that.

Thx again and just ping me once I could test something.

0xGiddi, 2024-10-29:
Agree with everything you said, I also want only some basic features, my personal needs were something like the following:
Disk LEDs:
 - Green to show disk is present
 - Blink to show activity
 - Blink for disk warnings (e.g. high temp warning, NVME/SSD life remaining/spare threshold)
 - Solid red for errors (ZFS data corruption detection, failing disk in SMART) etc.  
When I actually implement a userland script I'll add a documentations to what I use each LED for so other people may have the same idea/setup. 

I will keep the LEDs simple and allow the userland scripts to decide what to do with them. And I agree it will be up to the user to map the disk WWN/EUI (whatever by-id/by-path) to the correct LED. The LEDs will be named according to QNAPs original names ("Disk 1" "M.2 SSD 1" will  be "drivername:red:disk_1" and "drivername:green:m2ssd_1") I will not reindex disks from 0 or anything like that so that the hardware slot numbers correspond to numbers printed/etched onto the hardware (any disk layout will be the same as the QNAP user manual), names are derived from the disk sections in the model configuration file (See https://github.com/0xGiddi/tsx73a-ec/blob/master/dev_files/configs/model_Q07D0_Q07N0_10_10.conf for TS-473A example, "[System Disk 1]" is "M.2 SSD 1" and "[System Disk 3]" is actually "Disk 1").

> And I also had that problem, that 2 leds did nothing after a while..
> but without changing anything, last time when replaced a disk it worked.
Interesting, and this was two HDDs in two right slots?

> IDENT is not so important for me, because I have 1 NAS :)
Neither for me, but it might help someone with rackmount versions

 > In general I will not rely that much on the LED for monitoring the health or alerting, because I use other tools for that.
True, I get critical notifications via Telegram (Off topic, but if you are interested https://giddi.net/posts/monitoring-servers-using-telegram/), but having a bright LED flash when passing by the NAS is always a good idea.

jukrut, 2024-10-30:
>Interesting, and this was two HDDs in two right slots?

yes!
I did not think too much about it, because that are my 2 hdd (mirror) for backup data..

and my NAS is in another room, so I see the LED's rarely.. only when i want to change a disk or so.

just had a look now.
4 disks or in,
all led are green on.

but when i access with `dd` only disk 1 and 2 blink.

nvme led is also not blinking.

I already had a look and also found the post about the serial port :)

maybe also a "stealth" mode would be nice.. so no leds on at all.
(not a problem for me now, but blinking LEDs somehow disturbed me when i watched a movie or listened to music, but the HDD sound disturbed me even more, so i put it in another room.. )
that stealth mode could be controlled by global brightness correct?

1 thing i still don't understand..

what actually controls the LEDS currently and can that be disabled?
(i mean presence and blinking.)

0xGiddi, 2024-10-30:
> maybe also a "stealth" mode would be nice.. so no leds on at all. (not a problem for me now, but blinking LEDs somehow disturbed me when i watched a movie or listened to music, but the HDD sound [...]

Yes, I'll create a pseudo-LED that will control the brightness for all LEDs in the front panel (drivername::panel_brightness or similar), the individual  LED control is only a binary state, ON or OFF, but the EC does allow for a value that controls all the LEDs brightness at once (via internal PWM).

As for thr HDD noise, it's annoying when the HDD seek while reading/writing and I have no other proper space for mine but in the living-room, I found that most of the sound could be dampened by using screws on the HDD cages to hold the HDD properly in place and using some stiff foam squares on the 4 feet to stop vibrations resonating with the furniture it was sitting on. 

> 1 thing i still don't understand..
> 
> what actually controls the LEDS currently and can that be disabled? (i mean presence and blinking.)

I'm not 100% sure what controls  the activity LEDs at the moment, but as I mentioned above, its probably due to the two disks not being on a PCI bridge that allows for "standard" HDD strobe interfaces to be used, either directly by the SATA driver chip or standard EC/South-Bridge interfaces. 

I do believe from my tests (which were a while ago) that I found that setting the presence LED (solid-green) does override the default activity blinking. And it makes sense that it can be overridden since QNAP also want to set the error led state on it.

0xGiddi, 2024-11-07:
> I'm not 100% sure what controls the activity LEDs at the moment, but as I mentioned above, its probably due to the two disks not being on a PCI bridge that allows for "standard" HDD strobe [...]

Small update: 
Still working on the LEDs, but it seems that the green disk LEDs that blink with disk activity (without a special driver, out-of-the-box) seem to blink even if I disable the "activity" blink mode in the EC and even when the present (solid green) mode is set. Priority for the disk LEDs seems to be in this order:
1. Ident (blinking red) takes priority over all LED states
2. Error (solid red) takes priority over green LEDs
3. Present (solid green) only takes priority from activity (blinking green) when OFF (so disk activity will not cause the LED to turn on)
4. Activity (blinking green) takes priority over present (solid green), so disk activity causes the green LEDs to blink if its ON.

Problems:
1. I have not found a way to decouple the blinking of the two  diskLEDs that work out of the box, if error/ident is not set (red) they will blink with disk activity.
2. You cannot blink the two disk LEDs that work out of the box using the EC commands, the ONLY blink with disk activity. (the other disks that don't blink with activity you CAN blink with the EC command), you can also not stop the blinking if they are blinking from disk activity. 

This is annoying, as now some LEDs behave differently from others, and for other devices, it will be a manual process to figure out which LEDs can be blinked and which cant. 

I'm still trying to figure out the best way to set up the all the LED devices, on the one hand software blinking is nice (fast blink, slow blink, patterns) but on the other hand I would like to offload the work to the hardware (which means static blink with no speed control). But still, if HDD LEDs are set to blink and activity happens, it blinks OVER the software blink.

Plain annoying at this point.

0xGiddi, 2024-11-10:
OK, I have pushed to the master branch basic LED support (let call it "no blink support"). This creates LEDs for all disks (currently for TS-[4/6]73A) and for the status and USB. 

The disk/status LEDs are simple, they have 3 brightness settings:
0 - Off
1 - On GREEN
2 - On RED

The value USB LED only has 0 and 1 (OFF/ON) and the `panel-brightness` goes from 0 - 255. 

The disk LEDs blink on activity (for the disks that actually do, that I cannot control) when the LEDs are set to GREEN. Blink support is not really working properly yet and using `ledtrig_timer` module to blink them can work, but color selection is not handled properly yet.  I need to figure this out (again, maybe splitting all the LEDs into indevidual devices for each color, still the status LED is an outlier or keeping a "last color" cache in the custom led device wrapper). 

If you plan on blinking the LEDs manually or calling them on and off alot, you might want to consider removing `CFLAGS_tsx73a-ec.o := -DDEBUG` from `src/Makefile` to prevent dmesg spam. 

Yes, the code is a mess, names are all over the place and some duplicate calls can be minimized, but I wanted to push basic support for the LEDs.

jukrut, 2024-11-11:
Good Job!

I tested it shortly and it worked perfectly!

the most important tasks of leds (at least for me ):
- display error for drive
- display a general error like no network connection (status)
- identify drive

are doable and nothing crashes :)

I'll probably just run node-exporter locally and fetch the metrics, and decide based on them how the leds should behave.

so thank you!

jukrut, 2024-11-11:
the only tiny nitpick :)
i thing that number should be incremented
https://github.com/0xGiddi/tsx73a-ec/blob/master/dkms.conf#L2

0xGiddi, 2024-11-12:
Glad this small change work for you, I will be updating the driver now that I have properly started work on the LEDs (also, it really needs a good cleanup).

You are correct, the DMKS version should be incremented to prevent install errors (and the driver version itself should also be 0.2 not 1.0), I fixed the issue in the `v0.2-debug` tag and release. (I tagged my latest source, so some things with the fan and sensors detection in HWMON changed slightly to remove the model specific configuration, which is a WIP too, but this works on mine so should be fine with yours)
 
I will close this issue for now so bugs and issues could be tracked in new issues rather than one long thread. Fell free to open new issues for comments/bugs and requests as needed.

closed 2024-11-12 by 0xGiddi