0xGiddi/qnap8528 #3: TS464 IT8528 not found

adyanth, 2024-11-21:
I was trying to build and install this on TrueNAS. I am using a privileged ubuntu docker container to build and `insmod` this driver.

Once I build the `qnap8528.ko`, I try to `insmod qnap8528.ko`. This fails with 

```logs
$ make -C src all
$ insmod src/qnap8528.ko
insmod: ERROR: could not insert module src/qnap8528.ko: No such device
$ dmesg
[291468.952761] qnap8528 @ qnap8528_ec_hw_check: Could not locate IT8528 EC device
[291468.952777] qnap8528 @ qnap8528_init: qnap8528 device driver failed to probe, unloading
```

Does this mean the QNAP TS-464 does not use the embedded chip?

0xGiddi, 2024-11-21:
Can you please provide the following:
 - Do you have a more specific model number than TS-464 (any extra letters maybe on the label?)
 - Output of `/proc/ioports`
 - Output of the following program:
 ```c
#include <stdio.h>
#include <sys/io.h>
void main(void) {
        unsigned short ec_id;
        if (!ioperm(0x2e, 2, 1)) {
                outb(0x20, 0x2e);
                printf("First ID byte: 0x%x\n", inb(0x2f));
                outb(0x21, 0x2e);
                printf("Second ID byte: 0x%x\n", inb(0x2f));
                ioperm(0x2e, 2, 0);
        return;
    }
        printf("ioperm error\n");
}
```
(I should really print this in the logs when the error occurs, will add)

0xGiddi, 2024-11-21:
Also, if possible, can you statically compile the small program (if needed) and run it directly on the host without a container?

adyanth, 2024-11-21:
1. Not really, label says TS-x64, it is this (8G version): https://www.qnap.com/en-us/product/ts-464/specs/hardware
2.
```
$ cat /proc/ioports 
0000-0000 : PCI Bus 0000:00
  0000-0000 : dma1
  0000-0000 : pic1
  0000-0000 : timer0
  0000-0000 : timer1
  0000-0000 : keyboard
  0000-0000 : keyboard
  0000-0000 : rtc_cmos
    0000-0000 : rtc0
  0000-0000 : dma page reg
  0000-0000 : pic2
  0000-0000 : dma2
  0000-0000 : fpu
  0000-0000 : ene_ir
  0000-0000 : wdat_wdt
  0000-0000 : wdat_wdt
  0000-0000 : wdat_wdt
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
  0000-0000 : pnp 00:00
0000-0000 : PCI conf1
0000-0000 : PCI Bus 0000:00
  0000-0000 : pnp 00:01
  0000-0000 : ACPI PM1a_EVT_BLK
  0000-0000 : ACPI PM1a_CNT_BLK
  0000-0000 : ACPI PM_TMR
  0000-0000 : ACPI PM2_CNT_BLK
  0000-0000 : pnp 00:02
  0000-0000 : ACPI GPE0_BLK
  0000-0000 : pnp 00:06
  0000-0000 : PCI Bus 0000:06
  0000-0000 : PCI Bus 0000:05
  0000-0000 : PCI Bus 0000:04
  0000-0000 : PCI Bus 0000:03
  0000-0000 : PCI Bus 0000:02
  0000-0000 : PCI Bus 0000:01
  0000-0000 : 0000:00:02.0
  0000-0000 : 0000:00:1f.4
    0000-0000 : i801_smbus
```
3.
```
First ID byte: 0xff
Second ID byte: 0xff
```

I ran the same binary and got the same result as step 3 on the host (since host does not have gcc)

adyanth, 2024-11-21:
I also have the decompiled dsdt if that helps in any way.

[dsdt.dsl.txt](https://github.com/user-attachments/files/17840118/dsdt.dsl.txt)

0xGiddi, 2024-11-21:
Unfortunately, the output of /proc/ioports is missing, (all ranges are `0000`)  probably due to it running with insufficient permissions.

If the device uses the IT8528 the first and second ID bytes should be `0x85` and `0x28`. From the configuration I have for the TS-464, I cannot see that it uses any other chip than the IT8528. 

All these are the configuration files for various TS-464 devices, nothing else but IT8528:
```
./model_Q0930_10.conf:SIO_DEVICE = IT8528
./model_Q08S0_QY740_20_20.conf:SIO_DEVICE = IT8528
./model_Q07R1_Q08F0_20_11.conf:SIO_DEVICE = IT8528
./model_SAQ93_SBR00_10_10.conf:SIO_DEVICE = IT8528
./model_Q0910_Q08F0_21_11.conf:SIO_DEVICE = IT8528
./model_Q08S0_QY741_10_20.conf:SIO_DEVICE = IT8528
./model_Q07R6_Q08F0_21_11.conf:SIO_DEVICE = IT8528
./model_Q0910_Q08F0_20_11.conf:SIO_DEVICE = IT8528
./model_Q07R6_Q08F0_20_11.conf:SIO_DEVICE = IT8528
./model_Q07R1_Q08F0_13_11.conf:SIO_DEVICE = IT8528
./model_Q0930_20.conf:SIO_DEVICE = IT8528
./model_SAQ93_SBR00_13_10.conf:SIO_DEVICE = IT8528
./model_Q07R1_Q08F0_11_11.conf:SIO_DEVICE = IT8528
./model_Q0910_Q08F0_10_11.conf:SIO_DEVICE = IT8528
./model_Q07R6_Q08F0_11_11.conf:SIO_DEVICE = IT8528
./model_Q08S0_QY741_20_20.conf:SIO_DEVICE = IT8528
./model_Q08S0_QY740_10_20.conf:SIO_DEVICE = IT8528
./model_SAQ95_SBR00_12_11.conf:SIO_DEVICE = IT8528
```

It's not as "safe", but you can try to force load the module using `insmod qnap8528.ko skip_hw_check=true` skipping the hardware check for the chip. If the chip is working correctly, it would find a the boards model codes, and hopefully, wil match with one that exists, if it find the config (it will write it out to `dmsg`), then it's OK, and you can proceed to test the various controls to check that everything operates correctly. 

Please report back, and attach any output from the module so I can have it as a reference.

0xGiddi, 2024-11-21:
This is a bit silly, but I'm not sure where that return value is coming from. Can you edit the code and initialize `ret` in `qnap8528_probe` (line 1139) to 0? maybe it's a dirty page and the ret value is not 0-ed?

0xGiddi, 2024-11-21:
> This is a bit silly, but I'm not sure where that return value is coming from. Can you edit the code and initialize `ret` in `qnap8528_probe` (line 1139) to 0? maybe it's a dirty page and the ret value is not 0-ed? (`int ret = 0;` instead of `int ret;`)

Originally, this was  `ret = qnap8528_skip_hw_check ? 0 : qnap8528_ec_hw_check();` so I did not bother with 0ing `ret`

adyanth, 2024-11-21:
Same output . I replaced every single instance of `int ret;` with `int ret = 0;` and it still returned -524. I think it is the kernel returning [`-ENOTSUPP`](https://elixir.bootlin.com/linux/v4.9/source/include/linux/errno.h#L25) for some operation.

```
[296115.217163] qnap8528 @ qnap8528_probe: Skipping HW check for IT8528
[296115.905936] qnap8528: probe of qnap8528 failed with error -524
[296115.905977] qnap8528 @ qnap8528_init: qnap8528 device driver failed to probe, unloading
````=

adyanth, 2024-11-21:
They are indeed empty.

```
[296893.352168] qnap8528 @ qnap8528_probe: Skipping HW check for IT8528
[296894.028743] qnap8528 @ qnap8528_find_config: MB Code: 
[296894.028749] qnap8528 @ qnap8528_find_config: BP Code: 
[296894.028752] qnap8528: probe of qnap8528 failed with error -524
```

adyanth, 2024-11-21:
If you think it will help, I can try live booting ubuntu and running this kernel module to see if this is something to do with trueNAS  as a last resort (tomorrow maybe), although trueNAS scale is also based on Debian.

0xGiddi, 2024-11-21:
Hmmm... they should not be empty. 

I was thinking the same thing about reboot into a known system and then with direct access on the host test fore more possibilities. I will also write later a small C program that will dump the VPD tables, see if we can get any data out of them. I will post it here, but it will not be in the next few hours.

adyanth, 2024-11-21:
I printed the return value that sets those buffers and we are hitting this: https://github.com/0xGiddi/qnap8528/blob/master/src/qnap8528.c#L464 and getting `-EBUSY`. Do you know what it is trying to write and where? 

I ask because `EBUSY` is `Mount device busy` and my host OS has its root FS mounted as RO. I could remount necessary portions as RW if that is the issue.

0xGiddi, 2024-11-21:
Its breaking up the VPD entry value into the table/offset/length components (See https://github.com/0xGiddi/qnap8528/blob/v1.1/src/qnap8528.h#L27 ) and sending them to the EC for reading, it's not using a filesystem at all, all `qnap8528_ec_*` function are IO port operations using `inb` and `outb`. 

If its getting EBUSY, it might  mean that the read operations are stuck at waiting for the input buffer or output buffer of the chip to empty (which is either not happening, or not happening in time).

You can try and change the delay to 1000 for example and the retries to 5000 to see if its a timeout issue in (See https://github.com/0xGiddi/qnap8528/blob/v1.1/src/qnap8528.h#L27 ) 

Also, if you reboot, try going into the BIOS and checking if you can see information about the EC, it usually is displayed there (at least a version number)

0xGiddi, 2024-11-21:
In addition to making the retries 5000 and the sleep 1000, you might want to try this:

```
if (qnap8528_ec_write(reg_a, (offs >> 8) & 0xff) || qnap8528_ec_write(reg_b, offs & 0xff)) {
	udelay(5000);
        if(qnap8528_ec_read(reg_c, &raw[i])) {
		return -EBUSY;
        }
}
```

Here 
https://github.com/0xGiddi/qnap8528/blob/5a38cc3fc96975a28742ab9028d5d749f22a8815/src/qnap8528.c#L463

0xGiddi, 2024-11-21:
Try this branch https://github.com/0xGiddi/qnap8528/tree/ts-464 it only has the delay updates

adyanth, 2024-11-21:
That is crazy because it works now! The delay was all it needed (and skipping the hardware check, I wonder if there needs to be a delay for that too somewhere). A huge thanks to you for making this!

```
[322764.036133] qnap8528 @ qnap8528_probe: Skipping HW check for IT8528
[322765.075840] qnap8528 @ qnap8528_find_config: Searching configs for a match with MB=70-0Q07R1130 BP=70-1Q08F0110
[322765.075851] qnap8528 @ qnap8528_find_config: Model codes match found, model is TS-464
[322765.075918] input: qnap8528 as /devices/platform/qnap8528/input/input6
[322765.075982] qnap8528 @ qnap8528_register_inputs: Buttons input device registered
[322765.076071] qnap8528 @ qnap8528_register_leds: LED devices registered
[322765.171199] qnap8528 @ qnap8528_register_hwmon: Hwmon device registered
```

Here are some data I was exploring, let me know if you would like to see something specific. 
* Only the BTN_2 works (which is what I need). I dont think this has a chassis intrusion/open sensor, but I assumed the power button would be RESET? Or is there a pinhole reset button? I thought so because the power button also does not get reported with `/dev/input/event{0,1,2}`.
* I also noticed the temperature values seem arbitrary?
* The status LED status was incorrect on initialization. The LED was on and green, but the brightness reported 0. Changing it makes it work though, so probably not a big deal. Same with panel brightness, reported 0 when it was at a 100 initially.
* Setting `blink_bicolor` to 0 does not turn it off, instead the brightness needs to be set again.
* How do I blink a regular LED without bicolor, when the readme table says blink supported, see notes? I did not see a note for it. (Minor note, that table has incorrect name for the last entry for panel brightness as USB)
* HDD LEDs don't seem to be working correctly. They were working fine, but the brightness reported 0. I set it to 1, and LED changed color to red (was green and blinking with activity initially). Now, whatever I set the brightness to, it is stuck at red. I tested with hdd4 LED, I only have 4 drives.
* I also have a network LED that blinks with network activity, but that is not exposed. Not sure if it is tied directly to the NIC, and not a big deal regardless.

<summary>Random data</summary>
<details>

VPD:
```bash
root@qnas:/sys/devices/platform/qnap8528/vpd# ls | xargs -I{} bash -c 'echo {}; cat {}; echo'
backplane_date
2023-11-29 14:07:00
backplane_manufacturer
QNAP Systems
backplane_model
70-1Q08F0110
backplane_name
LF-SATA-BP
backplane_serial
72311015743
backplane_vendor
QNAP Systems
enclosure_nickname

enclosure_serial
Q23BF095953
mainboard_date

mainboard_manufacturer
QNAP Systems
mainboard_model
70-0Q07R1130
mainboard_name
STAT-6G-MB
mainboard_serial
72311014208
mainboard_vendor
QNAP Systems
```

EuP
```bash
root@qnas:/sys/devices/platform/qnap8528/ec# ls
cpld_version  fw_version
root@qnas:/sys/devices/platform/qnap8528/ec# cat cpld_version ; echo
0x2
root@qnas:/sys/devices/platform/qnap8528/ec# cat fw_version ; echo
Q07RL008
```

Buttons info
```bash
root@qnas:/sys/devices/platform/qnap8528/input/input6/capabilities# ls | xargs -I{} bash -c 'echo {}; cat {}'
[... 116 lines cut ...]
1
# ^ This actually turned the LED on and off
root@qnas:/sys/class/leds# cat qnap8528\:\:status/brightness 
0
root@qnas:/sys/class/leds# echo 2 > qnap8528\:\:status/brightness 
root@qnas:/sys/class/leds# echo 1 > qnap8528\:\:status/brightness 
root@qnas:/sys/class/leds# echo 0 > qnap8528\:\:status/brightness 
root@qnas:/sys/class/leds# echo 1 > qnap8528\:\:status/brightness 
# ^ Changes color correctly to green/red
root@qnas:/sys/class/leds# cat qnap8528\:\:panel_brightness/brightness 
0
root@qnas:/sys/class/leds# echo 50 > qnap8528\:\:panel_brightness/brightness 
root@qnas:/sys/class/leds# echo 10 > qnap8528\:\:panel_brightness/brightness 
root@qnas:/sys/class/leds# echo 100 > qnap8528\:\:panel_brightness/brightness 
root@qnas:/sys/class/leds# cat qnap8528\:\:hdd4/brightness 
0
root@qnas:/sys/class/leds# echo 1 > qnap8528\:\:hdd4/brightness 
root@qnas:/sys/class/leds# echo 2 > qnap8528\:\:hdd4/brightness 
root@qnas:/sys/class/leds# echo 3 > qnap8528\:\:hdd4/brightness 
root@qnas:/sys/class/leds# echo 0 > qnap8528\:\:hdd4/brightness 
root@qnas:/sys/class/leds# echo 0 > qnap8528\:\:hdd4/brightness 
# ^ Stuck at RED now, even 1 set it to red.
```

</details>

0xGiddi, 2024-11-21:
>That is crazy because it works now! The delay was all it needed (and skipping the hardware check, I wonder if there needs to be a delay for that too somewhere). A huge thanks to you for making this!

Amazing, Did you add any additional delays like I commented or did you use the ts-464 branch? Just so I know what exact delays and where worked. 

>Notes from my testing:
>* Only the BTN_2 works (which is what I need). I dont think this has a chassis intrusion/open sensor, but I assumed the power button would be RESET? Or is there a pinhole reset button? I thought so because the power button also does not get reported with `/dev/input/event{0,1,2}`.

Reset is the pinhole button (on my model), I have not checked if power button is reported at all.

>* I also noticed the temperature values seem arbitrary?

This might happen, as they are enumerated dynamically (any channel with a temperature between `0` and `128` is valid), if you check the README, you will see that temp channels `11` and `12` are for redundant power temp sensors, so they are invalid, but reporting a state I cannot discriminate against.  Also, be wary of `temp1`, as this is supposed to be "CPU", but CPU is temp reported by other modules (such as `k10temp` on my AMD device, and `temp1` does not match k10 temp reports) 

> * The status LED status was incorrect on initialization. The LED was on and green, but the brightness reported 0. Changing it makes it work though, so probably not a big deal. Same with panel [...]

Yes, the brightness reported back from the kernel is entirely software based (part of the led subsystem), so initially they will all be 0. 

>* How do I blink a regular LED without bicolor, when the readme table says blink supported, see notes? I did not see a note for it. (Minor note, that table has incorrect name for the last entry for panel brightness as USB)

LED blinking is done with standard Linux triggers (`echo <timer|oneshot> > /sys/class/leds/qnap8528\:\:status/trigger`), usually, this requires `ledtrig_timer`, `ledtrig_oneshot` or other kernel triggers  (`ledtrig_*` family) to be registered. This way, if hardware blink is not available the kernel will trigger the blinking via SW.

Maybe I should add a custom attribute to set the HW blink even if not triggers are available?

> * HDD LEDs don't seem to be working correctly. They were working fine, but the brightness reported 0. I set it to 1, and LED changed color to red (was green and blinking with activity initially). [...]

From the information you provided, your codes are `Q07R1` and `Q08F0`, which out of all the TS-464 devices, is the one that for some reason does not have EC indexes for setting the green LED, if that the case, the values `1` and `2` are treated as RED led. I thought that since I do not turn off the green LED (https://github.com/0xGiddi/qnap8528/blob/master/src/qnap8528.c#L611) it would revert to green when turning off red. 

The LEDs are a pain for the brain, currently, they only way to get them back would be to reset the NAS. I will push another change to TS-464 branch that will try and set the green  LED even if does not exist, maybe this is a mistake in QNAPs config I was parsing. 

>* I also have a network LED that blinks with network activity, but that is not exposed. Not sure if it is tied directly to the NIC, and not a big deal regardless.

There is no (known) interface for the network LED, it seems to be hardwired as far as I can tell. 

Thank you for the data much appreciated.

0xGiddi, 2024-11-21:
Please pull the latest TS-464 branch (https://github.com/0xGiddi/qnap8528/tree/ts-464) and check if you can set the LEDs to green.

0xGiddi, 2024-11-21:
Update: I think I found the bug:
https://github.com/0xGiddi/qnap8528/blob/ts-464/src/qnap8528.c#L613
I'm checking active and not error.  maybe the config change isn't needed. I'll revert the change to the config and fix the setting

0xGiddi, 2024-11-21:
Please pull https://github.com/0xGiddi/qnap8528/tree/ts-464 and try again

adyanth, 2024-11-21:
There was a hardware refresh beginning of 2024. That might be when it changed. I can try the changes later today and report back.

adyanth, 2024-11-22:
> Amazing, Did you add any additional delays like I commented or did you use the ts-464 branch? Just so I know what exact delays and where worked.

I was using the branch.

> LED blinking is done with standard Linux triggers 

Oh I see, that makes sense. I expected something else to be exposed since I saw the bicolor_blink. This is fine for me.

With v1.2:

It still does not pass the initial probe with:

```
[365596.139361] qnap8528 @ qnap8528_ec_hw_check: Could not locate IT8528 EC device
[365596.139376] qnap8528 @ qnap8528_init: qnap8528 device driver failed to probe, unloading
```

I have not restarted the NAS after the previous testing. Currently, echoing 0 to the HDD LED caused it to go back to like it was before (green and blinking) and 1 caused it to go red. I think that is how it is intended and works well.

Was there anything else you wanted me to test? Outside of needed to skip the hardware check, I am resolved with this issue. I'll leave the issue open to see if you needed any further follow ups/tests from me.

0xGiddi, 2024-11-22:
I did not remove the ec check from the  main branch, I will add more information in the supported models table and the install process for exceptions like the ts-464. 

Im happy everything is resolved, nothing else for now, I’ll keep the issue open just to remind me to go over everything and check that I have covered all things that came up either documentation. 

feel free to open another for any other problem and thank you for all the feedback.

closed 2024-11-23 by 0xGiddi

iwurui, 2025-07-26:
希望添加TS-564的支持。
dmesg：
[ 2866.020987] qnap8528 @ qnap8528_probe: Skipping HW check for IT8528
[ 2867.118988] qnap8528 @ qnap8528_find_config: Searching configs for a match with MB=70-0SAQ10120
[ 2867.118999] qnap8528 @ qnap8528_find_config: Could not find configuration for device, please report this issue
[ 2867.119027] qnap8528 qnap8528: probe with driver qnap8528 failed with error -524
[ 2867.119066] qnap8528 @ qnap8528_init: qnap8528 device driver failed to probe, unloading