0xGiddi/qnap8528 #4: Error when building on Truenas (I did run disable-rootfs-protection)

calypso, 2024-11-21:
Hardware: Qnap TS-873A
Software: Truenas Scale ElectricEel-24.10.0.2

Steps to reproduce
After disabling the rootfs protection I installed dkms then pulled the latest build. When compiling I get the following error.

`(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.44-production+truenas (x86_64)
Consult /var/lib/dkms/qnap8528/1.2/build/make.log for more information.
make: *** [Makefile:13: install] Error 10
`
/var/lib/dkms/qnap8528/1.2/build/make.log outputs this

`DKMS make.log for qnap8528-1.2 for kernel 6.6.44-production+truenas (x86_64)
Thu Nov 21 13:34:35 EST 2024
make[1]: Entering directory '/var/lib/dkms/qnap8528/1.2/build/src'
make -C /lib/modules/6.6.44-production+truenas/build M=$PWD modules
make[2]: Entering directory '/usr/src/linux-headers-6.6.44-production+truenas'
make[4]: *** [scripts/Makefile.build:243: /var/lib/dkms/qnap8528/1.2/build/src/qnap8528.o] Killed
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: /var/lib/dkms/qnap8528/1.2/build/src] Error 2
make[2]: *** [Makefile:237: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
make[1]: *** [Makefile:7: all] Error 2
make[1]: Leaving directory '/var/lib/dkms/qnap8528/1.2/build/src'
`
When I run dkms status however I see `qnap8528/1.2: added`

I then created the systemd entry and enabled it. I then received an error. systemctl status shows

`× qnap8528-load-module.service - Load qnap8528 EC kernel module
     Loaded: loaded (/etc/systemd/system/qnap8528-load-module.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-11-21 13:46:01 EST; 35s ago
    Process: 1821586 ExecStart=/sbin/modprobe qnap8528 (code=exited, status=1/FAILURE)
   Main PID: 1821586 (code=exited, status=1/FAILURE)
        CPU: 5ms

Nov 21 13:46:01 truenas systemd[1]: Starting qnap8528-load-module.service - Load qnap8528 EC kernel module...
Nov 21 13:46:01 truenas modprobe[1821586]: modprobe: FATAL: Module qnap8528 not found in directory /lib/modules/6.6.44-production+truenas
Nov 21 13:46:01 truenas systemd[1]: qnap8528-load-module.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 13:46:01 truenas systemd[1]: qnap8528-load-module.service: Failed with result 'exit-code'.
Nov 21 13:46:01 truenas systemd[1]: Failed to start qnap8528-load-module.service - Load qnap8528 EC kernel module.
`

Expected behavior:
The module installs without error and starts when enabling the systemctl service.

0xGiddi, 2024-11-21:
A you suggest, it seems that DKMS is working fine in registering the module, but the build process fails.

I am not well versed in the TrueNAS OS. Can you point me to the source for disabling rootFS protection (so I am at the same state as your machine)? I'll install TrueNAS this weekend and build it myself to see if I can trace the problem.

calypso, 2024-11-21:
Certainly.
the command I used is under /usr/local/libexec/disable-rootfs-protection. However I'm doubling back and I realized there is a command called install-dev-tools. I'm going to run that and see if that resolves the issue.

0xGiddi, 2024-11-21:
I had a TrueNAS VM laying around, I ran ` /usr/local/libexec/disable-rootfs-protection`, then `install-dev-tools` and `apt install dkms`. The kernel module compiled successfully.

I checked with `modprobe qnap8528` and it found the module and probed it (the probe failed do to it being a VM without an IT8528, but I got the error messages the module prints in the console).

calypso, 2024-11-21:
I attempted a reinstall of dkms and did another clone of the repo. Same errors

0xGiddi, 2024-11-21:
Can you please do the following:
1. Enter the downloaded source folder `cd qnap8528/src`
2. Edit the `/src/Makefile` and add `V=1` at the end of the line `$(MAKE) -C $(KERNEL_DIR) M=$$PWD modules
`
3. Try to compile manuall in the `src` directory with `make`

calypso, 2024-11-21:
Well some "good" news? The error changed slightly after that

```shell
cmd; rm -f /home/truenas_admin/qnap8528/src/.qnap8528.o.d"
make[3]: *** [scripts/Makefile.build:243: /home/truenas_admin/qnap8528/src/qnap8528.o] Killed
make[2]: *** [/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: /home/truenas_admin/qnap8528/src] Error 2
make[1]: *** [Makefile:237: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
make: *** [Makefile:7: all] Error 2
```

0xGiddi, 2024-11-21:
It seems that the problem is before the compilation of the actual module, when the compilation process is preparing.
1. Verify you are compiling as root (or `sudo make`)
2. In the same line above you added `V=1`, can you add `--trace`  after `$(MAKE)` (so its `$(MAKE) -C $(KERNEL_DIR) M=$$PWD modules `) and retry the `make` (please run `make clean` before running `make` again)

calypso, 2024-11-21:
Same error using both sudo make and adding trace. Should I remove V=1?

0xGiddi, 2024-11-21:
No need to remove `V=1`.

`--trace ` should provide a lot more than the small error above. It should trace every command executed in the build process. Something feels wrong specifically in your environment. 

Since the TrueNAS environment is usually stable, I could suggest a "hack" and to compile the kernel module in a fresh environment (on a virtual machine for example, I can provide you) and to copy the resulting `ko` over, but this does not fix your real environment, its just ignoring the symptoms. 

For context, this is the output from my test setup with `--trace` and `V=1`:
```
make --trace -C /lib/modules/6.6.44-production+truenas/build M=$PWD modules V=1
make[1]: Entering directory '/usr/src/linux-headers-6.6.44-production+truenas'
Makefile:237: target '__sub-make' does not exist
make --no-print-directory -C /usr/src/linux-headers-6.6.44-production+truenas \
-f /usr/src/linux-headers-6.6.44-production+truenas/Makefile modules
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:809: target 'include/config/auto.conf' does not exist
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1797: target 'prepare' does not exist
if [ "gcc (Debian 12.2.0-14) 12.2.0" != "gcc (Debian 12.2.0-14) 12.2.0" ]; then \
	echo >&2 "warning: the compiler differs from the one used to build the kernel"; \
	echo >&2 "  The kernel was built by: gcc (Debian 12.2.0-14) 12.2.0"; \
	echo >&2 "  You are using:           gcc (Debian 12.2.0-14) 12.2.0"; \
fi
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: update target '/root/qnap8528/src' due to: prepare
make -f ./scripts/Makefile.build obj=/root/qnap8528/src need-builtin=1 need-modorder=1 
scripts/Makefile.build:243: update target '/root/qnap8528/src/qnap8528.o' due to: /root/qnap8528/src/qnap8528.c FORCE
set -e; echo '# CC [M]  /root/qnap8528/src/qnap8528.o'; echo '  gcc -Wp,-MMD,/root/qnap8528/src/.qnap8528.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Werror -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g  -DMODULE  -DKBUILD_BASENAME='\''"qnap8528"'\'' -DKBUILD_MODNAME='\''"qnap8528"'\'' -D__KBUILD_MODNAME=kmod_qnap8528 -c -o /root/qnap8528/src/qnap8528.o /root/qnap8528/src/qnap8528.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --hacks=skylake --ibt --orc --retpoline --rethunk --sls --static-call --uaccess --prefix=16  --link  --module /root/qnap8528/src/qnap8528.o';   trap 'rm -f /root/qnap8528/src/qnap8528.o; trap - HUP; kill -s HUP $$' HUP;  trap 'rm -f /root/qnap8528/src/qnap8528.o; trap - INT; kill -s INT $$' INT;  trap 'rm -f /root/qnap8528/src/qnap8528.o; trap - QUIT; kill -s QUIT $$' QUIT;  trap 'rm -f /root/qnap8528/src/qnap8528.o; trap - TERM; kill -s TERM $$' TERM;  trap 'rm -f /root/qnap8528/src/qnap8528.o; trap - PIPE; kill -s PIPE $$' PIPE; gcc -Wp,-MMD,/root/qnap8528/src/.qnap8528.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Werror -std=gnu11 -fshort-wchar -funsigned-char -fno-common - [... cut ...]
  if [ ! -f vmlinux ]; then printf "Skipping BTF generation for %s due to unavailability of vmlinux
" /root/qnap8528/src/qnap8528.ko 1>&2; else LLVM_OBJCOPY="objcopy" pahole -J  --btf_base vmlinux /root/qnap8528/src/qnap8528.ko; ./tools/bpf/resolve_btfids/resolve_btfids -b vmlinux /root/qnap8528/src/qnap8528.ko; fi;
scripts/Makefile.modfinal:19: update target '__modfinal' due to: /root/qnap8528/src/qnap8528.ko
:
Makefile:233: update target 'modules' due to: __sub-make
:
make[1]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
```

calypso, 2024-11-21:
Wow...yea that is definitely not what I'm seeing lol.

0xGiddi, 2024-11-21:
I installed TrueNAS on my laptop under VMWare Workstation (paid software), you an use VirtualBox or any other software.
My config is as follows:
 - 2 vCPU
 - 4GB RAM
 - Bridged networking
 - 1 disk of 128GB

I installed TrueNAS on the single disk and then in the console (option 7) (not the webshell) ran  What i wrote in comment https://github.com/0xGiddi/qnap8528/issues/4#issuecomment-2492182518

Ill try and understand the error you are getting, it might take me some time.

calypso, 2024-11-21:
> Ill try and understand the error you are getting, it might take me some time.

No worries. I totally understand. I'm just super happy for help in general :)

I'll work on building the VM and I'll get back to you.

calypso, 2024-11-21:
So an update: I reran make but I pushed as much as I could into a txt file with sudo make > output.txt > 2>&1

and I got this

```shell
make --trace -C /lib/modules/6.6.44-production+truenas/build M=$PWD modules V=1
make[1]: Entering directory '/usr/src/linux-headers-6.6.44-production+truenas'
Makefile:237: target '__sub-make' does not exist
make --no-print-directory -C /usr/src/linux-headers-6.6.44-production+truenas \
-f /usr/src/linux-headers-6.6.44-production+truenas/Makefile modules
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:809: target 'include/config/auto.conf' does not exist
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (                \
echo >&2;                                                       \
echo >&2 "  ERROR: Kernel configuration is invalid.";           \
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";      \
echo >&2 ;                                                      \
/bin/false)
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1797: target 'prepare' does not exist
if [ "gcc (Debian 12.2.0-14) 12.2.0" != "gcc (Debian 12.2.0-14) 12.2.0" ]; then \
        echo >&2 "warning: the compiler differs from the one used to build the kernel"; \
        echo >&2 "  The kernel was built by: gcc (Debian 12.2.0-14) 12.2.0"; \
        echo >&2 "  You are using:           gcc (Debian 12.2.0-14) 12.2.0"; \
fi
/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: update target '/home/truenas_admin/qnap8528/src' due to: prepare
make -f ./scripts/Makefile.build obj=/home/truenas_admin/qnap8528/src need-builtin=1 need-modorder=1
scripts/Makefile.build:243: update target '/home/truenas_admin/qnap8528/src/qnap8528.o' due to: /home/truenas_admin/qnap8528/src/qnap8528.c FORCE
set -e; echo '# CC [M]  /home/truenas_admin/qnap8528/src/qnap8528.o'; echo '  gcc -Wp,-MMD,/home/truenas_admin/qnap8528/src/.qnap8528.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/>
make[3]: *** [scripts/Makefile.build:243: /home/truenas_admin/qnap8528/src/qnap8528.o] Killed
make[2]: *** [/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: /home/truenas_admin/qnap8528/src] Error 2
make[1]: *** [Makefile:237: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
make: *** [Makefile:7: all] Error 2
```

0xGiddi, 2024-11-21:
So, after reading through the over and over, the line ‘ make[3]: *** [scripts/Makefile.build:243: /home/truenas_admin/qnap8528/src/qnap8528.o] Killed’ does bot right with me. There should be an error if its a build sustem issue, not just ‘killed’.  It could be a permission issue, but we already checked that. Could this be the OOM killer? Could you check that there is enough free memory when compiling? Also, check ’dmesg’ after compiling to see if OOM killer was called? (Look for ‘oom’, ‘killed’ ‘Out of memory’ and similar)

calypso, 2024-11-21:
I sure hope not with 64GB of RAM with no apps currently running lol. I just ran the command again and checked dmesg and I don't see any OOM errors.

0xGiddi, 2024-11-21:
It was worth a shot, I cannot figure out why it would be killed and not exiting with an error condition. I don't know enough about TrueNAS architecture if it might be something else, I did read many complaints about the OOM killer getting invoked in 24.04.x.x for other processes. 

I think this is an environmental issue rather than a build system issue that is specifically with your setup. You can try and reboot if its not too much of a convenience just to have a “fresh” system. 

I would start digging through logs, monitoring the system load and memory and more while trying. but alas, I dont own the system so I’m afraid without new information its hard ti help ither that suggest more and more debugging steps. 

Compiling on an external machine like a VM is still an option. If i can think if anything I’ll comment.

calypso, 2024-11-21:
> You can try and reboot if its not too much of an inconvenience just to have a “fresh” system.

Thats the plan after the scrub finishes tomorrow lol (literally just got out of an abusive relationship with QuTS Hero 3 days ago lol) It would be really funny if it turns out the solution was just rebooting.

rwz, 2024-11-22:
Hi, just wanted to pop in and mention that I'm experiencing the same issue with my TS-673A

Independently tried some of the suggested remediation steps before discovering this thread.

0xGiddi, 2024-11-22:
@rwz  Thank you for replying this, it's helps to know this is not an isolated case. Can you confirm make is also being killed? Not just an error? 

 I will install TrueNAS on my TS-473A see if it happens to me and debug from there. The motherboard of all 3 devices is the same, so if it's somehow the architecture I might catch it. I'll update one I have some findings.

calypso, 2024-11-22:
I'm honestly just happy its not an isolated case. I wasn't sure if it was something I had done or if I was doing something wrong

0xGiddi, 2024-11-22:
Mixed news everybody! I have managed to kind of reproduce the error myself. It appears that I did not encounter this problem when I tested it since I was using TrueNAS console, not the web shell or SSH. I have just installed TrueNAS 24.10.0.2 on a spare NVME drive o my TS-473A and just like your experience, when compiling via the webshell the process was killed (after a lot of output). I also enabled SSH and tried to compile it there, but it was killed as well. I think the problem might be `sudo` no sure yet.

There are two solutions currently that work for me (assuming you have run `install-dev-tools`) here is my step by step:

**First solution - DKMS:**
Install either using `make install` in the root project directory or build only the module using `make` in the `src` directory. I did not encounter any problems with this.

**Second solution - NO DKMS:**
Since you can run `dkms` from `truenas_admin` without sudo (and the makefile `dkms`), download and compile the module only (`make` in  `src/` directory) using the `truenas_admin` user (no `sudo`!) and then use `sudo insmod` on the resulting `qnap8528.ko` manually. In this method, you will need to manually add the kernel module to the system so `modprobe` will find it.

I am working on figuring out why running under `sudo` is problematic.

calypso, 2024-11-22:
When running make without sudo

```shell
In file included from ./arch/x86/include/asm/linkage.h:6,
                 from ./include/linux/linkage.h:8,
                 from ./arch/x86/include/asm/cache.h:5,
                 from ./include/linux/cache.h:6,
                 from ./arch/x86/include/asm/current.h:9,
                 from ./include/linux/sched.h:12,
                 from ./include/linux/delay.h:23,
                 from /home/truenas_admin/qnap8528/src/qnap8528.c:30:
./arch/x86/include/asm/ibt.h:77:1: error: 'nocf_check' attribute ignored. Use '-fcf-protection' option to enable it [-Werror=attributes]
   77 | extern __noendbr u64 ibt_save(bool disable);
      | ^~~~~~
./arch/x86/include/asm/ibt.h:78:1: error: 'nocf_check' attribute ignored. Use '-fcf-protection' option to enable it [-Werror=attributes]
   78 | extern __noendbr void ibt_restore(u64 save);
      | ^~~~~~
In file included from ./include/linux/string.h:294,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/paravirt.h:17,
                 from ./arch/x86/include/asm/irqflags.h:60,
                 from ./include/linux/irqflags.h:17,
                 from ./include/linux/rcupdate.h:26,
                 from ./include/linux/rculist.h:11,
                 from ./include/linux/pid.h:5,
                 from ./include/linux/sched.h:14:
In function 'strncpy',
    inlined from 'qnap8528_vpd_parse' at /home/truenas_admin/qnap8528/src/qnap8528.c:481:3,
    inlined from 'qnap8528_vpd_attr_show' at /home/truenas_admin/qnap8528/src/qnap8528.c:470:9:
./include/linux/fortify-string.h:81:33: error: '__builtin_strncpy' output may be truncated copying between 0 and 255 bytes from a string of length 255 [-Werror=stringop-truncation]
   81 | #define __underlying_strncpy    __builtin_strncpy
      |                                 ^
./include/linux/fortify-string.h:165:16: note: in expansion of macro '__underlying_strncpy'
  165 |         return __underlying_strncpy(p, q, size);
      |                ^~~~~~~~~~~~~~~~~~~~
/home/truenas_admin/qnap8528/src/qnap8528.c: At top level:
/home/truenas_admin/qnap8528/src/qnap8528.c:1228:1: fatal error: opening dependency file /home/truenas_admin/qnap8528/src/.qnap8528.o.d: Permission denied
 1228 | module_exit(qnap8528_exit);
      | ^~~~~~~~~~~
cc1: all warnings being treated as errors
compilation terminated.
make[3]: *** [scripts/Makefile.build:243: /home/truenas_admin/qnap8528/src/qnap8528.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.44-production+truenas/Makefile:1924: /home/truenas_admin/qnap8528/src] Error 2
make[1]: *** [Makefile:237: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
make: *** [Makefile:7: all] Error 2
truenas_admin@truenas[~/qnap8528/src]$ ls
Makefile  output.txt  qnap8528.c  qnap8528.h```

Still an error but hopefully another datapoint?

calypso, 2024-11-22:
> `sudo insmod` on the resulting `qnap8528.ko` manually. In this method, you will need to manually add the kernel module to the system so `modprobe` will find it.

After doing sudo insmod is there anything else we need to do?

0xGiddi, 2024-11-22:
You might want to copy the module over to `/lib/modules/<kernel version>` and run `sudo depmod -a`. 
Unload the kernel module if you inserted it manually using `rmmod qnap8528` and then, run `modprobe qnap8528` (run it from a directory where the `.ko` is not there to just be sure). Check `dmsg` that the module probed and all functions work as you need and want. 

You can install the `systemd` script to load it on boot,  however, I have not tested this yet with TrueNAS.

0xGiddi, 2024-11-22:
> If a process can be created then a process can be automated :)

As long as that process does not need `sudo`. 

If everything is ok and working as expected feel. free to close the issue and open any other ones if needed.

closed 2024-11-22 by calypso