0xGiddi/qnap8528 #5: Suggestion: Script templates or links to them

calypso, 2024-11-23:
While I realize the code itself doesn't control the actual fans I did do a cursory check for fan scripts and as someone not well versed I'm not sure what I found would work. Is there any way in the README to either add a skeleton script for fan control or if you have a preferred script from another dev to link that? If not it's totally fine just thought it might be helpful to myself and others.

Thank you again for all the work and effort you put into this project!

0xGiddi, 2024-11-23:
Take a look at fancontrol and pwmconfig. If I remember correctly sometimes pwmconfig has problems automatically detecting the fans due to slow spin down/spin up but it can bet set manually.

 https://linux.die.net/man/8/fancontrol

 I will look into adding a section about this to the readme. A bit hard to do from my phone.

calypso, 2024-11-23:
Totally understand. Just realized after I had gotten everything set up that I didn't actually have any scripts to use with it lol.

0xGiddi, 2024-11-24:
I have not had the time to add anything, but I did quickly look at my device to see the setup, I do have a configuration for fancontrol **I do not recommend this specifically, its an example of what I have, model specific**:
```
INTERVAL=5
DEVPATH=hwmon3=devices/pci0000:00/0000:00:01.7/0000:07:00.0/ata2/host1/target1:0:0/1:0:0:0 hwmon6=devices/pci0000:00/0000:00:18.3 hwmon7=devices/platform/qnap8528
DEVNAME=hwmon3=drivetemp hwmon6=k10temp hwmon7=qnap8528
FCTEMPS=hwmon7/pwm7=hwmon6/temp1_input hwmon7/pwm1=hwmon3/temp1_input
FCFANS=hwmon7/pwm7=hwmon7/fan7_input hwmon7/pwm1=hwmon7/fan1_input
MINTEMP=hwmon7/pwm7=20 hwmon7/pwm1=20
MAXTEMP=hwmon7/pwm7=60 hwmon7/pwm1=50
MINSTART=hwmon7/pwm7=40 hwmon7/pwm1=40
MINSTOP=hwmon7/pwm7=0 hwmon7/pwm1=40
MINPWM=hwmon7/pwm1=0
```

I seem to have tested in the past all my fan speeds vs PWM, as I have graphs for them in my notes (again, model speific!):

[image]
[image]

I'm using the temperature of the CPU to control `fan7` which is my CPU side fan and the temperature of the disk that gets the hottest in the drive bays to control the disk fans. I do have a note in my "TODO" list to write a better script that will monitor an array of sensors, since (from what I've written down 8 months ago ) fancontrol cannot use multiple sensors to trigger a single fan (this might be wrong). If I ever get to writing that script, I'll let you know...

Sorry I could not help you more, I'm sure there are better resources out there tan me for this kind of stuff. As I said, I will look into adding a section on this once I research the topic.

calypso, 2024-11-25:
Awesome! Thank you so much. I apologize for the noob questions this is just the first time I've dealt with fan control in particular. So the primary reason I'm asking is because I have a secondary PCIe qnap card that has a 10gig connection and two nvme m.2 adapters. This has a fan and I'm trying to make sure that fan is either being picked up or is being dynamically adjusted by the card. I also apologize for asking all the questions. This is my first time really needing to mess with this and hardware to software interfacing has never been particularly strong for me lol. This is my output when I check sensors. 

``` shell
Devices:
hwmon0 is enp4s0
hwmon1 is nvme
hwmon2 is nvme
hwmon3 is nvme
hwmon4 is nvme
hwmon5 is k10temp
hwmon6 is qnap8528

Current temperature readings are as follows:
hwmon0/temp1_input      76
hwmon0/temp2_input      76
hwmon1/temp1_input      52
hwmon1/temp2_input      52
hwmon1/temp3_input      49
hwmon2/temp1_input      48
hwmon2/temp2_input      48
hwmon2/temp3_input      47
hwmon3/temp1_input      43
hwmon4/temp1_input      48
hwmon5/temp1_input      84
hwmon6/temp1_input      55
hwmon6/temp6_input      40
hwmon6/temp7_input      38
hwmon6/temp8_input      54
```
When I check my fancontrol configuration there is nothing there and I'm not sure you got the devpath values.

0xGiddi, 2024-11-25:
FYI just adding a picture, the marked circles are the two  temperature sensors I have found of the board, as you can see they are close to the CPU heatsink and the NVME drive (no installed here). I think the third sensor is hidden behind the heatsink of the CPU, and as I wrote in the README, I have no idea what is `temp1` hooked up to... it ight be a false or double reading. 
[image]

0xGiddi, 2024-11-25:
I think the QM2 does indeed have code to handle external QNAP expansions, I'll need to dig in more into this, can you confirm please the expansion cards you have if they are QNAP or not? If you can link to the product page, better if you have an exact model, as there a a few types of QM2 device for example. 

I will look into adding support for these devices, however,  it will probably require significant time and will not be a quick weekend project (also, you might need to test thinks and report back)

calypso, 2024-11-25:
> What cards are you using?

QNAP QM2-2P10G1TB: It's a combo card. It has both a 10Gbe RJ45 and dual m.2 NVME. The other card is an Nvidia P2000 lol.

> (if I remember correctly you have the TS-673A),

Close :) TS-873A (8 drive bays but I believe its the identical motherboard)

> except, my temps sensors are much lower

I also have heatsinks on all of my nvme's. It's likely due to the NAS's present location where its being suffocated (weekend project to fix this upcoming. I just didn't have the time lol)

> can you attach the qnap8528 sensors output?:

sure thing!
```shell
qnap8528-isa-0000
Adapter: ISA adapter
fan1:        1802 RPM
fan2:        1850 RPM
fan7:        4813 RPM
temp1:        +39.0°C
temp6:        +29.0°C
temp7:        +28.0°C
temp8:        +39.0°C

nvme-pci-0900
Adapter: PCI adapter
Composite:    +38.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

nvme-pci-0500
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

enp4s0-pci-0400
Adapter: PCI adapter
PHY Temperature:  +67.0°C
MAC Temperature:  +67.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +59.0°C

nvme-pci-0800
Adapter: PCI adapter
Composite:    +34.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

nvme-pci-0600
Adapter: PCI adapter
Composite:    +52.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +52.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +38.9°C  (low  = -273.1°C, high = +65261.8°C)
```

> but I manually copied the script and changed the delays in it to 30 seconds each

I honestly think my qnap, or at least the nvme card would literally cook if I did that lol.

> Can you also please verify that the device has 2 original fans (except for the PSU?) a smaller CPU side fan and a larger disks fan?

There are three fans built into the NAS not including the power supply and one fan on the 10g/nvme card.
2x 120mm Chasis fans
1x 60mm "CPU" fan (If that's a cpu fan I'm angelina jolie)

> I bought these (https://www.amazon.com/dp/B082VTVP51?ref=ppx_yo2ov_dt_b_fed_asin_title)

Okay wow. Those are the literal same ones I purchased and put on my drives 2 weeks ago. Crazy. 

> If you can link to the product page, better if you have an exact model, as there a a few types of QM2 device for example.

Here ya go! https://www.qnap.com/en-us/product/qm2-2p10g1tb

> I will look into adding support for these devices, however, it will probably require significant time and will not be a quick weekend project (also, you might need to test thinks and report back)

Hey I'm always up for tinkering as long as it doesn't nuke my data ;)

0xGiddi, 2024-11-25:
About the pwmconfig delays, you can play with the script and change it where you need, all you really want is for it to create a mapping between the hwmon devices. 

Makes sense you have 3 fans for the 873, I will read through the disassembly and find out to to figure out your system, and other QM2 expansion cards and if I can integrate them, I might post here some code I will need you to run to fill in missing information as I see there is an eeprom/PIC on the card that has information in it, it should not affect data or anything as we are still only probing the EC and the card info, but I cannot make that a guarantee, I hope you understand. 

I would of gladly run the code on my own device if I had any QM2 hardware, but I don't, and currently 200 USD is a bit steep for a pet project (maybe if I find a deal or something).

Since we are getting off topic from the original post, I think we need to move QM2 support to a new issue, and keep this one for the fans if you need any more info, I'll create an issue and tag you on it

calypso, 2024-11-26:
> About the pwmconfig delays, you can play with the script and change it where you need, all you really want is for it to create a mapping between the hwmon devices.

So my base assumption has always been since I have an odd number of fans, the two that run at relatively the same speed are the case fans and the one outlier is the 'cpu' fan. Which would likely mean fan 1 and 2 are the case fans and fan 7 is cpu. My issue though is that your script has additional data (specifically this 'devices/pci0000:00/0000:00:01.7/0000:07:00.0/ata2/host1/target1:0:0/1:0:0:0')
and I'm not exactly sure where and how to find it lol.

0xGiddi, 2024-11-26:
Yes, PWM fans are banked (see https://github.com/0xGiddi/qnap8528?tab=readme-ov-file#fan-reportingcontrol-and-temperature-sensors) so in your case fan1/2 are in group 1 "System fans" , and fan 7 is under "CPU fans" in the original config.

`devices/pci0000:00/0000:00:01.7/0000:07:00.0/ata2/host1/target1:0:0/1:0:0:0` is reading the temperature of the HDDs in the 3.5" bays, using the `drivetemp` kernel module and I'm using that to control the fan for the disks (a single fan in the 473A). 

Here is my hwmon directory: 

```shell
0 drwxr-xr-x  2 root root 0 Nov 24 17:23 .
0 drwxr-xr-x 60 root root 0 Nov 24 17:23 ..
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon0 -> ../../devices/pci0000:00/0000:00:01.3/0000:03:00.0/nvme/nvme0/hwmon0
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon1 -> ../../devices/pci0000:00/0000:00:01.4/0000:04:00.0/nvme/nvme1/hwmon1
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon2 -> ../../devices/pci0000:00/0000:00:01.7/0000:07:00.0/ata1/host0/target0:0:0/0:0:0:0/hwmon/hwmon2
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon3 -> ../../devices/pci0000:00/0000:00:01.7/0000:07:00.0/ata2/host1/target1:0:0/1:0:0:0/hwmon/hwmon3
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon4 -> ../../devices/pci0000:00/0000:00:08.2/0000:09:00.0/ata6/host5/target5:0:0/5:0:0:0/hwmon/hwmon4
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon5 -> ../../devices/pci0000:00/0000:00:08.2/0000:09:00.0/ata7/host6/target6:0:0/6:0:0:0/hwmon/hwmon5
0 lrwxrwxrwx  1 root root 0 Nov 24 17:23 hwmon6 -> ../../devices/pci0000:00/0000:00:18.3/hwmon/hwmon6
0 lrwxrwxrwx  1 root root 0 Nov 24 17:32 hwmon7 -> ../../devices/platform/qnap8528/hwmon/hwmon7
```
0/1 - NVME Disks
2-5 - HDD temps from `drivetemp` module
6 - AMD CPU temp from `k10temp`
7 - `qnap8528` module sensors/fans

PCI paths may differ by you, especially since you have a different backplane and have extra PCI cards.