0xGiddi/qnap8528 #6: Feature Request: Add support for EC controlled expansion cards/units

0xGiddi, 2024-11-25:
Add enumeration and control/monitoring for expansion cards and slots such as the QM2 series which have EC controlled fans/LEDs/sensors. 

CC: @calypso

0xGiddi, 2024-11-26:
@calypso Any chance I can get an `lspci -v`?

calypso, 2024-11-26:
> @calypso Any chance I can get an `lspci -v`?

coming right up!

```shell
root@truenas[/home/truenas_admin]# lspci -v > lspci.txt
root@truenas[/home/truenas_admin]# nano lspci.txt
root@truenas[/home/truenas_admin]# cat lspci.txt
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Root Complex
        Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Root Complex
        Flags: fast devsel

00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 IOMMU
        Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 IOMMU
        Flags: bus master, fast devsel, latency 0, IRQ 25
        Capabilities: [40] Secure device <?>
        Capabilities: [64] MSI: Enable+ Count=1/4 Maskable- 64bit+
        Capabilities: [74] HyperTransport: MSI Mapping Enable+ Fixed+

00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
        Flags: fast devsel, IOMMU group 0

00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0] (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
        Flags: bus master, fast devsel, latency 0, IRQ 26, IOMMU group 1
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: f000-ffff [size=4K] [16-bit]
        Memory behind bridge: f5000000-f60fffff [size=17M] [32-bit]
        Prefetchable memory behind bridge: e0000000-f1ffffff [size=288M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
        Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [270] Secondary PCI Express
        Capabilities: [2a0] Access Control Services
        Capabilities: [370] L1 PM Substates
        Kernel driver in use: pcieport

00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0] (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
        Flags: bus master, fast devsel, latency 0, IRQ 27, IOMMU group 2
        Bus: primary=00, secondary=02, subordinate=07, sec-latency=0
        I/O behind bridge: 1000-1fff [size=4K] [16-bit]
        Memory behind bridge: f6400000-f6bfffff [size=8M] [32-bit]
        Prefetchable memory behind bridge: f2000000-f21fffff [size=2M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
        Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [270] Secondary PCI Express
        Capabilities: [2a0] Access Control Services
        Capabilities: [370] L1 PM Substates
        Kernel driver in use: pcieport

00:01.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0] (prog-if 00 [Normal decode])
        Subsystem: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
        Flags: bus master, fast devsel, latency 0, IRQ 28, IOMMU group 3
        Bus: primary=00, secondary=08, subordinate=08, sec-latency=0
        I/O behind bridge: 2000-2fff [size=4K] [16-bit]
        Memory behind bridge: f7800000-f78fffff [size=1M] [32-bit]
        Prefetchable memory behind bridge: f2200000-f23fffff [size=2M] [32-bit]
        Capabilities: [50] Power Management version 3
        Capabilities: [58] Express Root Port (Slot+), MSI 00
        Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
[... 516 lines cut ...]
        Capabilities: [64] Express Endpoint, MSI 00
        Capabilities: [a0] MSI: Enable- Count=1/2 Maskable- 64bit+
        Capabilities: [c0] MSI-X: Enable+ Count=2 Masked-
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [2a0] Access Control Services
        Kernel driver in use: i2c_amd_mp2
        Kernel modules: i2c_amd_mp2_pci

0e:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 61) (prog-if 01 [AHCI 1.0])
        Subsystem: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode]
        Flags: bus master, fast devsel, latency 0, IRQ 148, IOMMU group 31
        Memory at f7500000 (32-bit, non-prefetchable) [size=2K]
        Capabilities: [48] Vendor Specific Information: Len=08 <?>
        Capabilities: [50] Power Management version 3
        Capabilities: [64] Express Endpoint, MSI 00
        Capabilities: [a0] MSI: Enable+ Count=2/2 Maskable- 64bit+
        Capabilities: [d0] SATA HBA v1.0
        Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
        Capabilities: [150] Advanced Error Reporting
        Capabilities: [270] Secondary PCI Express
        Capabilities: [2a0] Access Control Services
        Kernel driver in use: ahci
        Kernel modules: ahci

```

0xGiddi, 2025-01-08:
Hi @calypso can you check out the code I posted on #8 and run it youself? I'm trying to see if I can get a hack for tm QM2 cards as it came up again. 

https://github.com/0xGiddi/qnap8528/issues/8#issuecomment-2578406647

calypso, 2025-01-16:
Will do. Sorry for the delay. Was out of town.

0xGiddi, 2025-01-16:
No problem. Though from other users test it seems it did not work,  as did get another request for support of QM2 cards, I'm in the process of looking to purchase a chep one for development.

0xGiddi, 2025-01-16:
@nader-eloshaiker I have taken a look again into the disassembly of the QNAP binary I used to understand the control logic for the EC,  I do not see how the functions `ec_sys_<get/set>_fan_speed` can control QM2 expansion cards as fan IDs for the expansion cards (usually `1` and `2`) conflict with system fans. 

Taking a look at the QNAP-EC code you said worked for you, I see they use the exact same functions to get/set the fans speed and do not use the dedicated `ec_qm2_*` functions. @nader-eloshaiker can you confirm that you can actually control the QM2 fan speed using the QNAP-EC driver? I have a pesky feeling that the fan you are seeing and think that belongs to the QM2 is a phantom and not a real fan that is not reporting real values or cannot actually be controlled (or this may be a happy accident and the phantom PWM control groups aligned?)

As I mentioned in the README:
> Why fans are no enumerated at runtime:
I experimented with enumerating the fans at runtime by using a combination of checking the fan status in the fan status EC register, checking that the PWM values are between 0 and 255 and that the reported RPM is not a junk value (such as 65535 on my device), however, the RPM check is not enough and the junk value is not always a MAX_SHORT or something nice that can be detected. This method of enumeration also extends the module load. If this feature is requested it will not be hard to add, please create an issue requesting it.

And looking at the QNAP-EC code to understand why the discrepancy (https://github.com/Stonyx/QNAP-EC/blob/main/qnap-ec.c#L623) it seems they have the same logic (which I know from testing is a flawed)

@calypso Have you tried the QNAP-EC driver? How did that work for your QM2 cards?

0xGiddi, 2025-01-16:
This is `qm2_sys_get_fan_speed`

[image]

I saw this before, but I'm thinking that maybe more QM2 cards are controlled by I2C rather than the EC...I say this since looking at the EC QM2 functions, I do not see a way to "set" the expansion card index for reading, and the output of the code I provided in https://github.com/0xGiddi/qnap8528/issues/8#issuecomment-2578406647  should have produced something 
and not  https://github.com/0xGiddi/qnap8528/issues/8#issuecomment-2578984873

0xGiddi, 2025-01-18:
Progress update: I have ordered a test card to confirm my suspicions (it will take some time as it will arrive when I'm not around), after some more research into the code, the suspicion that the most QM2 card are not controlled by the EC is growing. This might be out of the scope of the qnap8528 driver, since, this is not part of the IT8528 EC (or ENE equivalent). 

Sine I have ordered the card, If I do manage to get the I2C bus working maybe I'll start a sister project for support of QM2 cards.

0xGiddi, 2025-01-19:
In order to try and prepare as much information before the card I ordered arrives, can you all run the following? I need as many samples as possible...

1. `for gpio in /sys/class/gpio/gpio*; do echo "GPIO: ${gpio##*/} - Label: $(cat $gpio/label 2>/dev/null || echo "No label")"; done`
2.  `lspci -nn -v`?
3. `ls -lash /sys/bus/i2c/devices/`
4. `for i2c in /sys/bus/i2c/devices/i2c-*; do echo "I2C: ${i2c##*/} - Name: $(cat $i2c/name 2>/dev/null || echo "No Name")"; done`
5. If you have `i2c-tools` installed, I'll appreciate also `i2cdetect  -l`

Thank you.

nader-eloshaiker, 2025-01-20:
Ok so here are a few things:

For point 1 -  here is the contents of the directory "/system/class/gpio"

drwxr-xr-x  2 root    0 Jan 20 14:39 ./
drwxr-xr-x 64 root    0 Jan 20 14:39 ../
--w-------  1 root 4096 Jan 20 20:57 export
lrwxrwxrwx  1 root    0 Jan 20 14:39 gpiochip512 -> ../../devices/platform/INT3453:00/gpio/gpiochip512/
lrwxrwxrwx  1 root    0 Jan 20 14:39 gpiochip592 -> ../../devices/platform/INT3453:01/gpio/gpiochip592/
lrwxrwxrwx  1 root    0 Jan 20 14:39 gpiochip672 -> ../../devices/platform/INT3453:02/gpio/gpiochip672/
lrwxrwxrwx  1 root    0 Jan 20 14:39 gpiochip692 -> ../../devices/platform/INT3453:03/gpio/gpiochip692/
--w-------  1 root 4096 Jan 20 20:57 unexport

Was I suppose to create the `gpio` file in this folder and chmod it to make it executable?

For point 2 - Here is the output

***@***.***[~]$ lspci -nn -v
00:00.0 Host bridge [0600]: Intel Corporation Gemini Lake Host Bridge [8086:31f0] (rev 06)
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Gemini Lake Host Bridge [8086:7270]
	Flags: bus master, fast devsel, latency 0, IOMMU group 1

00:02.0 VGA compatible controller [0300]: Intel Corporation GeminiLake [UHD Graphics 600] [8086:3185] (rev 06) (prog-if 00 [VGA controller])
	DeviceName: Onboard - Video
	Subsystem: Intel Corporation GeminiLake [UHD Graphics 600] [8086:2212]
	Flags: bus master, fast devsel, latency 0, IRQ 150, IOMMU group 0
	Memory at a0000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 90000000 (64-bit, prefetchable) [size=256M]
	I/O ports at f000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: <access denied>
	Kernel driver in use: i915
	Kernel modules: i915

00:0e.0 Multimedia audio controller [0401]: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio [8086:3198] (rev 06)
	DeviceName: Onboard - Sound
	Subsystem: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio [8086:7270]
	Flags: bus master, fast devsel, latency 0, IRQ 25, IOMMU group 2
	Memory at a1710000 (64-bit, non-prefetchable) [size=16K]
	Memory at a1000000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_skl, snd_soc_avs, snd_sof_pci_intel_apl

00:0f.0 Communication controller [0780]: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface [8086:319a] (rev 06)
	DeviceName: Onboard - Other
	Subsystem: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface [8086:7270]
	Flags: bus master, fast devsel, latency 0, IRQ 149, IOMMU group 3
	Memory at a171b000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel driver in use: mei_me
	Kernel modules: mei_me

00:12.0 SATA controller [0106]: Intel Corporation Celeron/Pentium Silver Processor SATA Controller [8086:31e3] (rev 06) (prog-if 01 [AHCI 1.0])
	DeviceName: Onboard - SATA
	Subsystem: Intel Corporation Celeron/Pentium Silver Processor SATA Controller [8086:7270]
	Flags: bus master, 66MHz, medium devsel, latency 0, IRQ 133, IOMMU group 4
	Memory at a1714000 (32-bit, non-prefetchable) [size=8K]
	Memory at a171a000 (32-bit, non-prefetchable) [size=256]
	I/O ports at f090 [size=8]
	I/O ports at f080 [size=4]
	I/O ports at f060 [size=32]
	Memory at a1719000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: <access denied>
	Kernel driver in use: ahci
	Kernel modules: ahci

00:13.0 PCI bridge [0604]: Intel Corporation Gemini Lake PCI Express Root Port [8086:31d8] (rev f6) (prog-if 00 [Normal decode])
	Subsystem: Intel Corporation Gemini Lake PCI Express Root Port [8086:7270]
[... 119 lines cut ...]
	Kernel driver in use: igc
	Kernel modules: igc

For point 3 - here is the output

***@***.***[~]$ ls -lash /sys/bus/i2c/devices/
total 0
0 drwxr-xr-x 2 root root 0 Jan 20 14:39 .
0 drwxr-xr-x 4 root root 0 Jan 20 14:39 ..
0 lrwxrwxrwx 1 root root 0 Jan 20 21:03 0-0036 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-0/0-0036
0 lrwxrwxrwx 1 root root 0 Jan 20 21:03 0-0037 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-0/0-0037
0 lrwxrwxrwx 1 root root 0 Jan 20 14:39 0-0050 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-0/0-0050
0 lrwxrwxrwx 1 root root 0 Jan 20 14:39 i2c-0 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-0
0 lrwxrwxrwx 1 root root 0 Jan 20 21:03 i2c-1 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-1
0 lrwxrwxrwx 1 root root 0 Jan 20 21:03 i2c-2 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-2
0 lrwxrwxrwx 1 root root 0 Jan 20 21:03 i2c-3 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-3

For point 4 - here is the output

I2C:  - Name: No Name

Did you want me to install QNAP-EC to run this test or your qnap8528 (which is what I have at the moment)? Also, with regards to I2C-Tool, is there any specific instructions you want me to follow?

Regards
Nader

0xGiddi, 2025-01-20:
Thank you for the output, For point 1, you don't need to create a file, the script is just priting the label for each of the GPIO controllers on the system. for example, it does `cat /system/class/gpio/gpiochip512/label` and so on for each entry in `/sys/class/gpio`.

It does not matter what driver you have installed, the commands do not use any features from the driver to run the tests. About `i2c-tools`, I only need `sudo i2cdetect -l` output if you have the `i2c-tools` installed. 

As I look into this more and more, I do believe support for QM2 will be a separate driver as they are not controlled by the EC itself but by other interfaces and that no driver on the open market currently support control and feedback from QM2 devices. 

Thank you, I'll keep posting here as I get more in depth understanding.

0xGiddi, 2025-01-23:
After digging more into the QM2 code, it seems it is controlled using I2C /SMBus. In addition, the bus does not register automatically when QM2 PCI cards are added and requires a kernel driver to created the GPIO chip, I2C/SMBus device and remap PCI memory.  I'm in the process of reverse engineering, This will not be a quick task, but we are in the right direction.

0xGiddi, 2025-03-02:
Closing this issue as after research on the driver and QNAP app, it seems this QM2 are not controlled by the EC, hence, are no in the scope of this project.

 I have created a temporary repo that I'm adding *RAW* research into (it does not have functional code yet and current work is on HW i have on hand).  https://github.com/0xGiddi/qnap-qm2/tree/master