0xGiddi/qnap8528 #8: Device not found on TS-254D

nader-eloshaiker, 2025-01-08:
Followed the steps to install on truenas 24.10.1

- compiled, output below:

```
truenas_admin@truenas[/mnt/app-pool/bin/qnap8528/src]$ make
============================================
 If you are not me or a developer ignore;
 REMEMBER: All configuration entries with a
 MB code only should be at the end of the
 configuration array to avoid mixed models
============================================
make -C /lib/modules/6.6.44-production+truenas/build M=$PWD modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.44-production+truenas'
  CC [M]  /mnt/app-pool/bin/qnap8528/src/qnap8528.o
  MODPOST /mnt/app-pool/bin/qnap8528/src/Module.symvers
  CC [M]  /mnt/app-pool/bin/qnap8528/src/qnap8528.mod.o
  LD [M]  /mnt/app-pool/bin/qnap8528/src/qnap8528.ko
  BTF [M] /mnt/app-pool/bin/qnap8528/src/qnap8528.ko
Skipping BTF generation for /mnt/app-pool/bin/qnap8528/src/qnap8528.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.6.44-production+truenas'
```

- copied file using `cp qnap8528.ko /lib/modules/$(uname -r)/extra`
- ran `depmod -a` 
- generated `/etc/systemd/system/qnap8528-load-module.service`
- Running `systemctl start qnap8528-load-module.service` in a `su` session or as `sudo` gives an error:

```
Job for qnap8528-load-module.service failed because the control process exited with error code.
See "systemctl status qnap8528-load-module.service" and "journalctl -xeu qnap8528-load-module.service" for details.
```

- Running `systemctl status qnap8528-load-module.service` gives:

```
Jan 08 21:20:42 truenas systemd[1]: Starting qnap8528-load-module.service - Load qnap8528 EC kernel module...
Jan 08 21:20:43 truenas modprobe[11552]: modprobe: ERROR: could not insert 'qnap8528': No such device
Jan 08 21:20:43 truenas systemd[1]: qnap8528-load-module.service: Main process exited, code=exited, status=1/FAILURE
Jan 08 21:20:43 truenas systemd[1]: qnap8528-load-module.service: Failed with result 'exit-code'.
Jan 08 21:20:43 truenas systemd[1]: Failed to start qnap8528-load-module.service - Load qnap8528 EC kernel module.
Jan 08 21:20:43 truenas systemd[1]: qnap8528-load-module.service: Consumed 1.126s CPU time.
```

- `systemctl daemon-reload` had no effect
- adding `skip_hw_check=true` at the end of execstart had no effect

0xGiddi, 2025-01-08:
Could be I have not generated a configuration for the device, I'll look into this as soon as I can. Can you also post the lines from `dmesg` corresponding to the module insertion? I tried checking but it seems there is no TS-254D, maybe a typo and you meant TS-253D?

nader-eloshaiker, 2025-01-08:
[   86.171022] qnap8528 @ qnap8528_probe: Skipping HW check for IT8528
[   87.781948] qnap8528 @ qnap8528_find_config: Searching configs for a match with MB=70-0Q05H0201
[   87.781966] qnap8528 @ qnap8528_find_config: Could not find configuration for device, please report this issue
[   87.781977] qnap8528: probe of qnap8528 failed with error -524
[   87.782881] qnap8528 @ qnap8528_init: qnap8528 device driver failed to probe, unloading

nader-eloshaiker, 2025-01-08:
I should note that I also have a [Qnap QM2-2P-244A](https://www.qnap.com/en-au/product/qm2-2p-244a) installed in it too.

0xGiddi, 2025-01-08:
Please check the latest release/tag (v1.4) and check support. (Might still need to skip HW check, try without at first, checking `dmesg` please update me if it was required)

0xGiddi, 2025-01-08:
Also, Forgot to mention than fan reading/control and temperature readings from QM2 expansions cards (only the cards, not the whole NAs device with a card) is not currently supported due to lack of hardware on my end to run tests and development cycles.

nader-eloshaiker, 2025-01-08:
looks like it is working, just trying to get fancontrol working. I am happy to give you remote access to my truenas instance to test.

nader-eloshaiker, 2025-01-08:
I kind of really need to control the fan on the QM2, the HDD temps are good but the nvme's are too high, I have to spin up the main fun higher to cool them down to baseline.

0xGiddi, 2025-01-08:
There was a discussion about the QM2 cards inbthe past here. Unfortunately I do not have such a card so I'm unable to develop for that case. I have tried but without the card it's impossible to know what I'm doing. The card is pricy but I may get one in the future to try and develop support for that and other QM2 modules. 

If I have a good direction to try I'll try and update. Sorry this module does not cover that.

0xGiddi, 2025-01-08:
Can you please compile the following code and attach the results? 
```#include <sys/io.h>
#include <stdlib.h>
#include <stdio.h>
#include <unistd.h>

typedef unsigned char u8;
typedef unsigned short u16;

static int qnap8528_ec_wait_ibf_clear(void)
{
	int retries = 0;

	do {
		if (!(inb(0x6c) & 2))
			return 0;
		usleep(1000);
	} while (retries++ < 5000);
	return -1;
}

static int qnap8528_ec_clear_obf(void)
{
	int retries = 0;

	do {
		if (!(inb(0x6c) & 1))
			return 0;
		inb(0x68);
		usleep(1000);
	} while (retries++ < 5000);
	return -2;
}

static int qnap8528_ec_wait_obf_set(void)
{
	int retries = 0;

	do {
		if (inb(0x6c) & 1)
			return 0;
		usleep(1000);
	} while (retries++ < 5000);
	return -3;
}

static int qnap8528_ec_send_command(u16 command)
{
	int ret;

	ret = qnap8528_ec_wait_ibf_clear();
	if (ret)
		goto ec_send_cmd_out;
	outb(0x88, 0x6c);

	ret = qnap8528_ec_wait_ibf_clear();
	if (ret)
		goto ec_send_cmd_out;
	outb((command >> 8) & 0xff, 0x68);

	ret = qnap8528_ec_wait_ibf_clear();
	if (ret)
		goto ec_send_cmd_out;
	outb(command & 0xff, 0x68);

ec_send_cmd_out:
	return ret;
}

static int qnap8528_ec_read(u16 command, u8 *data)
{
	int ret;
	ret = qnap8528_ec_clear_obf();
	if (ret)
		goto ec_read_out;

	ret = qnap8528_ec_send_command(command);
	if (ret)
		goto ec_read_out;

	ret = qnap8528_ec_wait_obf_set();
	if (ret)
		goto ec_read_out;

	*data = inb(0x68);

ec_read_out:
	return ret;
}

static int qnap8528_ec_write(u16 command, u8 data)
{
	int ret;

	ret = qnap8528_ec_send_command(command | 0x8000);
	if (ret)
		goto qnap8528_ec_read_out;

	ret = qnap8528_ec_wait_ibf_clear();
	if (ret)
		goto qnap8528_ec_read_out;

	outb(data, 0x68);

qnap8528_ec_read_out:
	return ret;
}

void main(void) {
    u8 i, rpm, pwm;
    
    if (ioperm(0x6c, 1, 1) != 0) {
        perror("ioperm error on 0x6c");
        return;
    }

    if (ioperm(0x68, 1, 1) != 0) {
        perror("ioperm error on 0x68");
        return;
    }

    for (i=0; i<=64; i++) {
        qnap8528_ec_read(0x17d, &rpm);
        qnap8528_ec_read(0x17b, &pwm);
        printf("Reading QM2 fan %u, PWM=%u, RPM=%u (%u)\n", i, pwm, rpm, rpm * 0x3c);
    }

    ioperm(0x6c, 1, 0);
    ioperm(0x68, 1, 1);
    printf("Done\n");
}```

nader-eloshaiker, 2025-01-09:
I got the following error very quickly
```
ioperm error on 0x6c: Operation not permitted
```

0xGiddi, 2025-01-09:
Are you running it as root? If yes, try to unload the qnap8528 module.

nader-eloshaiker, 2025-01-09:
OK I had to run `sudo su` to make it work.

Here is the output:
```
Reading QM2 fan 0, PWM=255, RPM=255 (15300)
Reading QM2 fan 1, PWM=255, RPM=255 (15300)
Reading QM2 fan 2, PWM=255, RPM=255 (15300)
Reading QM2 fan 3, PWM=255, RPM=255 (15300)
Reading QM2 fan 4, PWM=255, RPM=255 (15300)
Reading QM2 fan 5, PWM=255, RPM=255 (15300)
Reading QM2 fan 6, PWM=255, RPM=255 (15300)
Reading QM2 fan 7, PWM=255, RPM=255 (15300)
Reading QM2 fan 8, PWM=255, RPM=255 (15300)
Reading QM2 fan 9, PWM=255, RPM=255 (15300)
Reading QM2 fan 10, PWM=255, RPM=255 (15300)
Reading QM2 fan 11, PWM=255, RPM=255 (15300)
Reading QM2 fan 12, PWM=255, RPM=255 (15300)
Reading QM2 fan 13, PWM=255, RPM=255 (15300)
Reading QM2 fan 14, PWM=255, RPM=255 (15300)
Reading QM2 fan 15, PWM=255, RPM=255 (15300)
Reading QM2 fan 16, PWM=255, RPM=255 (15300)
Reading QM2 fan 17, PWM=255, RPM=255 (15300)
Reading QM2 fan 18, PWM=255, RPM=255 (15300)
Reading QM2 fan 19, PWM=255, RPM=255 (15300)
Reading QM2 fan 20, PWM=255, RPM=255 (15300)
Reading QM2 fan 21, PWM=255, RPM=255 (15300)
Reading QM2 fan 22, PWM=255, RPM=255 (15300)
Reading QM2 fan 23, PWM=255, RPM=255 (15300)
Reading QM2 fan 24, PWM=255, RPM=255 (15300)
Reading QM2 fan 25, PWM=255, RPM=255 (15300)
Reading QM2 fan 26, PWM=255, RPM=255 (15300)
Reading QM2 fan 27, PWM=255, RPM=255 (15300)
Reading QM2 fan 28, PWM=255, RPM=255 (15300)
Reading QM2 fan 29, PWM=255, RPM=255 (15300)
Reading QM2 fan 30, PWM=255, RPM=255 (15300)
Reading QM2 fan 31, PWM=255, RPM=255 (15300)
Reading QM2 fan 32, PWM=255, RPM=255 (15300)
Reading QM2 fan 33, PWM=255, RPM=255 (15300)
Reading QM2 fan 34, PWM=255, RPM=255 (15300)
Reading QM2 fan 35, PWM=255, RPM=255 (15300)
Reading QM2 fan 36, PWM=255, RPM=255 (15300)
Reading QM2 fan 37, PWM=255, RPM=255 (15300)
Reading QM2 fan 38, PWM=255, RPM=255 (15300)
Reading QM2 fan 39, PWM=255, RPM=255 (15300)
Reading QM2 fan 40, PWM=255, RPM=255 (15300)
Reading QM2 fan 41, PWM=255, RPM=255 (15300)
Reading QM2 fan 42, PWM=255, RPM=255 (15300)
Reading QM2 fan 43, PWM=255, RPM=255 (15300)
Reading QM2 fan 44, PWM=255, RPM=255 (15300)
Reading QM2 fan 45, PWM=255, RPM=255 (15300)
Reading QM2 fan 46, PWM=255, RPM=255 (15300)
Reading QM2 fan 47, PWM=255, RPM=255 (15300)
Reading QM2 fan 48, PWM=255, RPM=255 (15300)
Reading QM2 fan 49, PWM=255, RPM=255 (15300)
Reading QM2 fan 50, PWM=255, RPM=255 (15300)
Reading QM2 fan 51, PWM=255, RPM=255 (15300)
Reading QM2 fan 52, PWM=255, RPM=255 (15300)
Reading QM2 fan 53, PWM=255, RPM=255 (15300)
Reading QM2 fan 54, PWM=255, RPM=255 (15300)
Reading QM2 fan 55, PWM=255, RPM=255 (15300)
Reading QM2 fan 56, PWM=255, RPM=255 (15300)
Reading QM2 fan 57, PWM=255, RPM=255 (15300)
Reading QM2 fan 58, PWM=255, RPM=255 (15300)
Reading QM2 fan 59, PWM=255, RPM=255 (15300)
Reading QM2 fan 60, PWM=255, RPM=255 (15300)
Reading QM2 fan 61, PWM=255, RPM=255 (15300)
Reading QM2 fan 62, PWM=255, RPM=255 (15300)
Reading QM2 fan 63, PWM=255, RPM=255 (15300)
Reading QM2 fan 64, PWM=255, RPM=255 (15300)
Done
```

0xGiddi, 2025-01-09:
Well. That's not the output I expected. I thought maybe one would show up properly and I could hack something together. It seems the QM2 controls are going to be a bit more tricky.

0xGiddi, 2025-01-09:
Apart from the QM2 expansion card control is there anything else not working properly that you noticed that needs fixing for this model? Otherwise all other comments about this should be in issue #6 dealing with expansion card support and it's off topic from this issue.

nader-eloshaiker, 2025-01-14:
Been running up for a few days now and it looks like it's working fine. I'll close the ticket. Shame about the QM2.

nader-eloshaiker, 2025-01-14:
Just a note, the QNAP-EC driver work with it.

closed 2025-01-14 by nader-eloshaiker

0xGiddi, 2025-01-14:
> Just a note, the QNAP-EC driver work with it.

I understand the frustration, the QNAP-EC driver only exposes a kernel interface to the proprietary QNAP binaries for hwmon , this means that although they have more fans support (though they do lack support for anything that is not a fan), they *legally* cannot publish a full solution (or it can be taken down by copyright claim) since they use QNAPs intellectual property (`libuLinux_hal.so)`, the 3d party binary).

One of goals of this project is to disconnect from QNAPs proprietary code/intellectual property by a process of reverse engineering the interface and re-implementing it, so a full open source driver for the EC is available for all, shielded from a potential copyright/IP takedown claim (or worse, legal prosecution) and to keep a small, clean codebase that maybe one day would be integrated into OSs as a standard driver, so we don't need to mess around with finding what works (and no OS project will include a driver using proprietary binaries). 

That is why it takes some time to gain more and more features (at first only fans, then EuP/Power recovery, LEDs and etc...). As more and more users request things, and hardware becomes available more features will be added and contributed. As I mentioned earlier, I simply do not have the hardware to test and develop with, I did some reverse engneering on the QM2 interface, but without hardware it's hard to tell if its working (like the short test I gave you earlier to test a theory that did not work out).

When I'll have some spare cash and time, I might go out and purchase a expansion card for future development...  
All the best!

nader-eloshaiker, 2025-01-15:
Can you start a go fund me, more than happy to donate.RegardsNaderSent from my ENIAC®

0xGiddi, 2025-01-15:
I appreciate the offer very much, but a little backstory;

I started this driver as a pet project to get basic support for my own device and did not plan on it being a "real" driver and to support multiple models and more (that's why it was named originally *tsx73a-ec*), it sat dormant for a year until a request to support more features was opened and only in November last year I reworked the code and smartened it up so it could be used properly by other people and device models. 

Since I work full time, I do not feel comfortable to take money for something I cannot promise any results or dedicated time to research and develop. I will look into buying an cheap QM2 expansion card to try and develop and add support for QM2 devices (at least to get the communication to the expansion card down). If I get it working, then I may consider to add a method for people to "reimburse" costs of  development hardware or donate to add support for specific hardware.

Thank you again for your offer, I will try to remember to mention you when I add support for QM2 devices.

Atomique13, 2025-01-15:
Just wanted to add that TS-253D works with "skip HW check" flag. I am running OMV if that helps. Great job with the qnap8528, works much better than the old qnap-ec !

0xGiddi, 2026-04-13:
Added to QM2 repo README. Thank you.