0xGiddi/qnap8528 #12: Help in trying to implement a one-touch-copy systemd service under Debian

christophehenry, 2025-02-25:
Hi there! Thank you for your work. I'm trying to implement implement a one-touch-copy systemd service on top of this code but this is really low-level and I lack the knwoledge to progess. I have two questions
1. You compilation process installs the module at `/lib/modules/$(uname -r)/extra` but, unless I'm wrong, that module will stop being loaded with each kernel change, unless recompiled?
2. How can I know what USB brigde is located next to the button and how can I link that bridge with a hypothetic plugged device? Does that bridge change with each Qnap model (I suppose so)?

0xGiddi, 2025-02-25:
You are correct when not using DKMS, this makes sense as if there is a change in the kernel API you do not want the old module to panic and crash the system (or worse, destroy something such as data). That's why I recommend using DKMS, as most package managers (verify this for your own) will trigger a DKMS rebuild and compile the module against the correct kernel headers version. (Ofc, this is not true for systems such as TrueNAS that replace the rootFS on upgrades, manual kenrel updates compiling from source, or "obscure" distros with custom package managers). 

I'm not in front of a computer but these are my thoughts on the USB:

**I.**  Bridges will change between QNAP models, especially if they do not share a motherboard model or if they have other hardware attached that has USB hubs on them (this might change kernel enumeration order).

**II.**  If the NAS does not have a block devices connected via USB, why care to which physical port they are connected? make it backup drives plugged into the back too... (if there are permanent USB block devices connected, maybe blacklisting the them in the backup script?)

**III.** Locating the correct physical USB port using bus/port numbers could be an option (these might change depending on USB version/speed used by the device (so USB 2 and USB 3 devices on the same port might appear on different buss/port numbers)). This will require some logic to find the correct block device associated with the bus/port. (`lsusb -t`, probing `/sys/bus/usb`, `/dev/disk/by-path` to figure this out)

**IV.** I would try and poll the `EV_KEY` + `BTN_2` (the copy button) event only while a SUB block device is connected (why poll the EC every 100ms if there is nothing to do?) 

**V.** What logic to use for handling multi partition devices (a HDD/flash drive with multiple partitions)

christophehenry, 2025-02-25:
So if I understand correctly, right now, there's no way to determine which where's the disk plugged into the USB port next to the button and the simplest solution is to just guess or copy drive plugged in USB?

0xGiddi, 2025-02-25:
Unfortunately, there is no magical command to determine the physical location of the port, that's just how the kernel works, there is the `physical_location` attribute group under `/sys/bus/usb/<bus>-<port>` and `/sys/bus/usb/<bus>-<port>/port`, but I have rarely seen this actually reflect the real world placement of the ports. 

The best thing to do if you want a physical location check for the USB port is to figure out a point in the device tree where anything that is connected after that point is probably connected to the front USB, such as a bus/port pair or if it's unique enough the PCI BDF of the USB controller.

christophehenry, 2025-02-25:
Yes, but `/sys/bus/usb/<bus>-<port>` is always populated in a reproductible way based on placement on motherboard, right? So if I determine that on my own NAS, the disk is currently mounted on `/sys/bus/usb/devices/4-0:1.0`, can I expect to it to always be located there and can I expect it to be located ther on every device of that model? If no, do you have an idea on how Qnap determines where to copy from?

0xGiddi, 2025-02-25:
The bus and port numbers should be predictable as long as there is no USB controller hardware changes (like adding a hub). They are generated dynamically but if the hardware does not change the enumeration order should stay the same (and there is something else in the kernel that should help but I won't go into it). 

However, it might change if the block device is using usb 2 or 3 (or 1). I have not played around with the usb subsystem in a while, so always double check.  Also you can try and plug it in and out, use different devices with different protocols and version and reboots and see the output of dmesg for consistency.

christophehenry, 2025-02-25:
Alright, thank you!

christophehenry, 2025-02-26:
Quick question: you own a TS-673A, right? Can you confirm me with dmesg that front USB port corresponds to USB 2-4 ?

0xGiddi, 2025-02-26:
I have a 473A, front USB is indeed 2-4 for a USB 3 flash drive I have plugged in. I'm not physically near the NAS right now to check for USB 2 or 1 devices.

0xGiddi, 2025-02-26:
SuperSpeed devices (USB 3/3.1) is on 2-4, HiSpeed and FullSpeed (USB 2/1.1) are on 1-4.

christophehenry, 2025-02-26:
Yep. I think I found a solution. Basically, devices déclare their physical location in `/sys/devices/.../physical_location` so I was able to group all the possible hubs by their physical location. I think it's reliable. I may share some Python code for that if you're interested to test?

0xGiddi, 2025-02-26:
Interesting, I found that physical location attributes cannot be trusted on my NAS (or laptop), all the ports have the same location (I have 2 devices plugged in the back (1-3/1-4) and a flash drive in the front (2-4))

```
root@QNAS-D11:/sys/bus/usb/devices# for d in [0-9]*-[0-9]*([0-9]); do echo "USB bus-port $d:"; for f in $d/physical_location/*; do printf "\t%s: %s\n" "$f" "$(cat "$f")"; done; done
USB bus-port 1-2:
        1-2/physical_location/dock: no
        1-2/physical_location/horizontal_position: left
        1-2/physical_location/lid: no
        1-2/physical_location/panel: left
        1-2/physical_location/vertical_position: lower
USB bus-port 1-3:
        1-3/physical_location/dock: no
        1-3/physical_location/horizontal_position: left
        1-3/physical_location/lid: no
        1-3/physical_location/panel: left
        1-3/physical_location/vertical_position: lower
USB bus-port 2-4:
        2-4/physical_location/dock: no
        2-4/physical_location/horizontal_position: left
        2-4/physical_location/lid: no
        2-4/physical_location/panel: left
        2-4/physical_location/vertical_position: lower
```

And sure, I don't mind testing (when I get the chance to be near the NAS), maybe if it all works we can even add it as an example usage of the driver if you will allow it :).

christophehenry, 2025-02-27:
You're right. USB2 and USB3 devices show up under different physical location here too.

christophehenry, 2025-02-28:
So I did a few more tests on my machine. Everything is detailed [in this stackoverflow question](https://stackoverflow.com/posts/79471983) but, long story short, on a TS-673A, it appears that a USB device plugged on the front will always show up under `Bus 01.Port 1-Port 4` for USB2 and `Bus 02.Port 1-Port 4` for USB3. Can you replicate those results?

0xGiddi, 2025-02-28:
Will do. Will only get around to in as Sunday evening for physical access.

christophehenry, 2025-02-28:
SO link fixed. So based on my findings, I wrote the following UDEV rules:

```udev
ACTION=="add", SUBSYSTEM=="block", KERNELS=="2-4", SUBSYSTEMS=="usb", SYMLINK+="qnap-one-touch-copy"
ACTION=="add", SUBSYSTEM=="block", KERNELS=="1-4", SUBSYSTEMS=="usb", SYMLINK+="qnap-one-touch-copy"
```

It should symlink the device to `/dev/qnap-one-touch-copy` when something is plugged on the front. I'd be glad if you could test this too.

0xGiddi, 2025-03-02:
Just checked the rules. I checked the front USB port as well as one of the ports on the back (I cannot unplug the device in one of the ports in the back). 

I tested using 4 device, A USB 3 flash drive, a USB 2 flash drive, a WiFi adapter and my UPS USB port. 

I can confirm that only when the USB 2 and USB 3 flash drives were connected the system created a the symlink `/dev/qnap-one-touch-copy` to the relevant block device.

From here there are things that need to be done: 
1. Decide on a backup strategy (sync changes, backup, backup location)
2. Currently, the udev rule symlinks the entire drive, so a strategy of how to handle multi-partitoned devices. Example, a HDD might have a UEFI partition (or other partitions), so just backing up the first partition might not be sufficient and wanted. backing up the entire block device will only result in an image, not a browsable copy. the backup script needs to be tailored to your choosing.
3. While the symlink is detected, a script to turn on the USB on the front to show it has detected the drive, maybe flash while copying (using ledtrig_timer module) and turning off on unmout.
4. Create a script to detect button press itself using `evtest`/python/similar so that the button can be used for example a short press to start backup, long press (lets say 5 seconds) to unmount the device and allow safe removeal.

I might work on a script like this in the near future (I dont really have a use for it, but I think it can be a neat example on how to use the module).

christophehenry, 2025-03-03:
Thank you for your inputs. I already have working code for detecting the drive, mounting partitions and handling button events. I think I will settle on calling rsync with `--progress` for copying since there doen't seem to be a copy function that reports progress in the standard library. I don't understand how to use the LEDs in Python though. Do you have an input for that?

Also, would you be interested in setting up an organisation for this repo and mine?

0xGiddi, 2025-03-03:
Very well, I would also probably have used `rsync` since it can be configured to copy in a verity of ways. 

The most basic way to work with the LED indicators (ON and OFF) is to just write the value `0` or `1` to `/sys/class/leds/qnap8528::usb/brightness`. This will turn the LED on and off. If you want more advanced such as blinking while copying, you can load the `ledtrig-timer` kernel module (it's usually built in most systems) and then set the `/sys/class/leds/qnap8528::usb/trigger` to `timer` (you can also change the frequency using `delay_on` and `delay_off`, but this will cause it to be software controlled, when done set `0` and trigger to `none`).

That way you can set the LEDs to indicate that a USB was detected (solid on), copying (blinking) and when done and unmounted, turn the LED off to indicate safe removal. 

I'm not sure I understood the meaning of this, ```Also, would you be interested in setting up an organisation for this repo and mine?```

christophehenry, 2025-03-03:
> I'm not sure I understood the meaning of this, Also, would you be interested in setting up an organisation for this repo and mine?

I mean, instead of hosting your repo and mine on our respective accounts, [create a free Github organisation](https://github.com/organizations/plan) for all the code related to implementing Qnap features under Linux.

christophehenry, 2025-03-03:
Alright [I pushed the code here](https://github.com/christophehenry/qnap-one-touch-copy-linux). This is a rough implementation but it's working with LED support. Next step I need to find a packaging system — probably [meson](https://mesonbuild.com/) to allow installation with your driver, a systemd service,  and the udev rule and also enable the kernel modules.

0xGiddi, 2025-03-04:
I'm sorry I have not tested this yet or have feedback. In the little free time I have from work, I have been tidying up other things. I will get around to it. I will also create the organization, I have reserved the name qnap8528, and I already have more repos I would like to add (some code I have had laying around and finally decided to give it a repo and a README and some research I'm doing into expansion cards).

https://github.com/0xGiddi/qnap8528-grub-status if this interests you. 

I promise, I will get around to it!

christophehenry, 2025-03-05:
It's ok, don't apologize. I understand the burden of developing free software during free time.

0xGiddi, 2025-03-06:
I'm trying to learn the Debian (and others) packaging systems myself in the past few weeks to try and created packages for this and other projects (unsatisfactory so far, there are to many ways and build systems). 

If I manage anything, I'll let you know.