0xGiddi/qnap8528 #14: issue on FnOS

gzxiexl, 2025-04-15:
FnOS is a new NAS operating system (https://fnnas.com/download) based on a customized Debian kernel. I tried installing it on a TS-453Bmini .
I followed the tutorials but couldn’t compile and install it properly.

```
root@TS453Bmini:/home/gzxiexl/qnap8528# make install
dkms install .
make: dkms: No such file or directory
make: *** [Makefile:13: install] Error 127
```

gzxiexl, 2025-04-15:
```
root@TS453Bmini:/home/gzxiexl/qnap8528# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
root@TS453Bmini:/home/gzxiexl/qnap8528# uname -a
Linux TS453Bmini 6.6.38-trim #92 SMP PREEMPT_DYNAMIC Tue Mar 11 17:22:50 CST 2025 x86_64 GNU/Linux
```
`root@TS453Bmini:/home/gzxiexl/qnap8528# sudo apt install -y dkms gcc make libssl-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
make is already the newest version (4.3-4.1).
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu cpp cpp-12 dpkg-dev fakeroot gcc-12 libalgorithm-diff-perl
  libalgorithm-diff-xs-perl libalgorithm-merge-perl libasan8 libatomic1 libbinutils libcc1-0 libctf-nobfd0 libctf0
  libdpkg-perl libfakeroot libfile-fcntllock-perl libgcc-12-dev libgprofng0 libisl23 libitm1 liblsan0 libmpc3 libmpfr6
  libssl3 libtsan2 libubsan1 linux-compiler-gcc-12-x86 linux-headers-6.1.0-33-amd64 linux-headers-6.1.0-33-common
  linux-headers-amd64 linux-kbuild-6.1 openssl
Suggested packages:
  binutils-doc cpp-doc gcc-12-locales cpp-12-doc menu debian-keyring gcc-multilib manpages-dev autoconf automake
  libtool flex bison gdb gcc-doc gcc-12-multilib gcc-12-doc bzr libssl-doc
Recommended packages:
  build-essential libc6-dev | libc-dev libc6-dev
The following NEW packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu cpp cpp-12 dkms dpkg-dev fakeroot gcc gcc-12
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl libasan8 libatomic1 libbinutils libcc1-0
  libctf-nobfd0 libctf0 libdpkg-perl libfakeroot libfile-fcntllock-perl libgcc-12-dev libgprofng0 libisl23 libitm1
  liblsan0 libmpc3 libmpfr6 libssl-dev libtsan2 libubsan1 linux-compiler-gcc-12-x86 linux-headers-6.1.0-33-amd64
  linux-headers-6.1.0-33-common linux-headers-amd64 linux-kbuild-6.1
The following packages will be upgraded:
  libssl3 openssl
2 upgraded, 37 newly installed, 0 to remove and 225 not upgraded.
Need to get 67.7 MB of archives.
After this operation, 262 MB of additional disk space will be used.
Get:1 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libisl23 amd64 0.25-1.1 [683 kB]
Get:2 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libmpfr6 amd64 4.2.0-1 [701 kB]
Get:3 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libmpc3 amd64 1.3.1-1 [51.5 kB]
Get:4 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 cpp-12 amd64 12.2.0-14 [9,764 kB]
Get:5 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 cpp amd64 4:12.2.0-3 [6,836 B]
Get:6 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libcc1-0 amd64 12.2.0-14 [41.7 kB]
Get:7 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 binutils-common amd64 2.40-2 [2,487 kB]
Get:8 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libbinutils amd64 2.40-2 [572 kB]
Get:9 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libctf-nobfd0 amd64 2.40-2 [153 kB]
Get:10 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libctf0 amd64 2.40-2 [89.8 kB]
Get:11 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libgprofng0 amd64 2.40-2 [812 kB]
Get:12 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 binutils-x86-64-linux-gnu amd64 2.40-2 [2,246 kB]
Get:13 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 binutils amd64 2.40-2 [65.0 kB]
Get:14 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libitm1 amd64 12.2.0-14 [26.1 kB]
Get:15 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libatomic1 amd64 12.2.0-14 [9,328 B]
Get:16 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libasan8 amd64 12.2.0-14 [2,195 kB]
Get:17 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 liblsan0 amd64 12.2.0-14 [969 kB]
Get:18 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libtsan2 amd64 12.2.0-14 [2,196 kB]
Get:19 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libubsan1 amd64 12.2.0-14 [883 kB]
Get:20 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libgcc-12-dev amd64 12.2.0-14 [2,437 kB]
Get:21 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 gcc-12 amd64 12.2.0-14 [19.3 MB]
Get:22 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 gcc amd64 4:12.2.0-3 [5,216 B]
Get:23 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libdpkg-perl all 1.21.22 [603 kB]
Get:24 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 dpkg-dev all 1.21.22 [1,353 kB]
Get:25 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 dkms all 3.0.10-8+deb12u1 [48.7 kB]
Get:26 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libfakeroot amd64 1.31-1.2 [28.3 kB]
Get:27 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 fakeroot amd64 1.31-1.2 [66.9 kB]
Get:28 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libalgorithm-diff-perl all 1.201-1 [43.3 kB]
Get:29 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libalgorithm-diff-xs-perl amd64 0.04-8+b1 [11.4 kB]
Get:30 https://mirrors.tuna.tsinghua.edu.cn/debian bookworm/main amd64 libalgorithm-merge-perl all 0.08-5 [11.8 kB]
[... cut ...]
make: *** [Makefile:13: install] Error 3
root@TS453Bmini:/home/gzxiexl/qnap8528# make clean
make: *** No rule to make target 'clean'.  Stop.
root@TS453Bmini:/home/gzxiexl/qnap8528# make install
dkms install .
Sign command: /lib/modules/6.6.38-trim/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Error! DKMS tree already contains: qnap8528-1.8
You cannot add the same module/version combo more than once.
make: *** [Makefile:13: install] Error 3`

gzxiexl, 2025-04-15:
Since the kernel of FnOS is a customized version, can I set up an x64 Linux virtual machine, compile the program there, and then copy it over to this NAS?

0xGiddi, 2025-04-15:
Can you please provide output for `dkms status`?

gzxiexl, 2025-04-15:
Because Fnos has a minimalist kernel and doesn't include compilation tools by default, I eventually set up a Docker compilation environment on the host machine to compile the qnap8528.ko file. After manually copying it out and loading it, I successfully obtained fan1.

[image]

0xGiddi, 2025-04-15:
Happy you got it working,

If you don't mind writing down the steps (in short), I'll try and recreate this and add a guide to the README file.
Also, if you can confirm it works I would add that it was tested on the README model list. 

Thank you.