0xGiddi/qnap8528 #16: 希望添加对TS-564的支持

iwurui, 2025-07-27:
我能通过https://github.com/Stonyx/QNAP-EC 驱动EC，EC实际型号是：ENE KB9068QB

0xGiddi, 2025-07-27:
1. Please provide a full model number of the device and a description of the problem.
2. Please provide output logs of qnap8528  from `dmsg` if applicable.

0xGiddi, 2025-07-27:
It seems to be that this model is not international and specific to the Chinese market and thus does not appear on qnap.com. I'll have to dig through the website you provided and get the firmware. I'll try and get around to it tonight and add support for this model.

iwurui, 2025-07-27:
它大概只在中国销售，配置类似于TS-464C2，不同之处：TS-564为SATA3.5寸盘位*3 + SATA2.5寸盘位*2。非常感激您的帮助！

0xGiddi, 2025-07-27:
Please check out release v1.10 and update if everything works:
https://github.com/0xGiddi/qnap8528/releases/tag/v1.10