0xGiddi/qnap8528 #20: Could not compile without modification of qnap8528.c

navierb, 2025-09-19:
Attempting to compile and install v1.13 on Debian 11 as per instructions with "make" throw some errors like:

`error: label at end of compound statement
    default:
    ^~~~~~~`

I had to add `break;` after lines 1077 and 1094 in the src/qnap8528.c file.

navierb, 2025-09-19:
It is running fine, it was only more of a comment for reference in case someone encounters the same problem.

Thank you very much for this project. Finally I can control the fans :-)

0xGiddi, 2025-09-19:
Fixed with tag 1.14 for compatibility with GCC 10 (https://github.com/0xGiddi/qnap8528/releases/tag/v1.14). 
Thank you for taking the time to raise the issue. If there are any other issues or requests, feel free to open a new one.

closed 2025-09-19 by 0xGiddi